motioneye-project/motioneye #216: Email notification logs

DawsonLeBrown, 2016-06-06:
Hi.

I've got motioneye running on the following platform:

Raspberry Pi 3
Raspbian - latest build
NoIR pi camera

I've turned on Send an Email under motion notifications, and added my gmail credentials, but am not receiving any emails. I've checked all of the log files I can find, but cannot find anything related to motioneye trying to send emails. Can you please let me know which log file I should look in - and possibly which log level i should use to be able to track down my issue.

Thanks,

Dawson

ccrisan, 2016-06-06:
What version of motionEye? Can you attach your `motioneye.log` file (`sudo journalctl -u motioneye > motioneye.log`)?

DawsonLeBrown, 2016-06-06:
Hi,

meyectl -v says it's 0.31.5.

I've attached the logfile.

[motioneye.log.txt](https://github.com/ccrisan/motioneye/files/300421/motioneye.log.txt)

Thanks

ccrisan, 2016-06-06:
One more thing, can you please set your log level to debug (in motioneye.conf), rerun everything and generate some motion so that it would normally try to send an email and recapture the motioneye.log file.

DawsonLeBrown, 2016-06-06:
Hi,

new log file attached.

[motioneye.log.txt](https://github.com/ccrisan/motioneye/files/300535/motioneye.log.txt)

Thanks

ccrisan, 2016-06-06:
Hmm everything looks good. If you manually execute the following command (taken from your log), does it work?

    /usr/bin/python /usr/local/lib/python2.7/dist-packages/motioneye/meyectl.pyc sendmail -c /etc/motioneye/motioneye.conf 'smtp.gmail.com' '587' 'changedemailaddress@gmail.com' 'NotMyPassword' 'True' 'changedemailaddress@gmail.com' 'changedemailaddress@gmail.com' 'motion_start' '1' '2016-06-06T15:44:45' '1'

Of course you'd need to change the values accordingly.

DawsonLeBrown, 2016-06-06:
It did not work, but from the logs that printed out I was able to identify that it was was just a simple username/password error.

Thanks for your help in debugging the problem, and also for the excellent work on Motioneye.

ccrisan, 2016-06-06:
@DawsonLeBrown that's great! I'll close the issue then.

closed 2016-06-06 by ccrisan