motioneye-project/motioneye #217: Feature request: pan-tilt control

yanoschik, 2016-06-06:
Great work!
And it would be awesome if there was a simple way to control a simple pan-tilt setup from the UI ...
Like using/interfacing [pi-blaster](https://github.com/sarfata/pi-blaster), [ServoBlaster](https://github.com/richardghirst/PiBits/tree/master/ServoBlaster), or similar. Here's one (outdated) example: [RPi_Cam_Web_Interface_ServoBlaster_pan_tilt](https://github.com/skalad/RPi_Cam_Web_Interface_ServoBlaster_pan_tilt).
Thanks

yanoschik, 2016-07-06:
I've created a _temporary_ solution ... a set of scripts which can be called by the action buttons.
See here: [PiPan Cam](https://github.com/yanoschik/PiPan_Cam)

ccrisan, 2016-07-31:
Zoom in/out buttons will also be included in the next release, coming out today.

closed 2016-07-31 by ccrisan

sfgabe, 2019-03-27:
If anyone is interested in another updated option, I've added my pan tilt scripts here: 
https://github.com/sfgabe/custom_scripts/tree/master/motioneye_scripts/pan-tilt

It's fully 3D printable (link in the readme) and uses the pigpio library, which I've found to be more reliable for cheap micro servos.

yo3hjv, 2019-05-20:
Hi Catalin, I am using MotionEye OS  20190427. Sorry to ask like an idiot: There is instructions for pan-tilt buttons somewhere?  
Multumesc,
Adrian

ccrisan, 2019-05-27:
@yo3hjv you won't get pan/tilt controls out-of-the-box with motionEye. But if you know how to control them via HTTP requests or any other programmable method, you can use [Action Buttons](https://github.com/ccrisan/motioneyeos/wiki/Action-Buttons) to implement such a functionality yourself.