motioneye-project/motioneye #219: Motioneye can't connect to raspberry pi motion camera

jcherndon, 2016-06-11:
I have a raspberry pi with motion installed and can verify works by going to ports 8080 and 8081.
I have motioneye 0.31.5 running on a centos7 box.
When I try to connect from motioneye to the raspberry pi i get a "not valid" error in the gui and a log of 
ERROR: failed to list remote cameras on http://192.168.2.7:8080: Not Valid
DEBUG: 200 GET /config/list/?_=1465662158611&scheme=http&host=192.168.2.7&port=8080&path=%2F&username=admin&password=&proto=motioneye&_username=admin&_signature=3a86ea4e4749cea1353f44a95288ecaf22217711 (192.168.2.184) 20.80ms

Ideas?

Oxize, 2016-06-12:
Did you select local camera and then the camera which is connected to the RPI?

ccrisan, 2016-06-12:
It looks like you're trying to add a remote motionEye camera but you provided the URL to a MJPEG stream. You'll probably want to specify port 80 (or none) instead of 8080.

jcherndon, 2016-06-12:
But the motion software on the pi is listening on 8080 and 8081 (verified with netstat).
Maybe I don't have a library installed on the pis? I was able to connect a pi installed with the motioneyeos to my motioneye (centos7) box.
I haven't done much with the motion.conf on the pis, just enabled remote admin and deamon processing.

jcherndon, 2016-06-12:
Do I need motioneye running on pis along with motion in order to connect with a different motioneye server?

jcherndon, 2016-06-12:
new to github, don't know why I closed this.

ccrisan, 2016-06-12:
I thought you were running motionEye on your PI as well. If you only run motion on your PI, then add the camera as a network camera (as opposed to Remote motionEye Camera).

jcherndon, 2016-06-12:
ok, that worked. Will I be able to control the motion (pi) though the motioneye (centos), i.e. change parameters and such?

ccrisan, 2016-06-12:
Nope.

jcherndon, 2016-06-12:
so then If I was to change motion parameters I'll need to install motioneye on the pis as well? Or is that something that doesn't get done?

Oxize, 2016-06-12:
Why dont you just access the PI from any box (either its windows, linux) Why you want within Centos from Motion to MotionEye PI?

ccrisan, 2016-06-12:
@jcherndon I thought that was obvious :)

jcherndon, 2016-06-12:
So maybe it'll help if I mentioned what I want to do.
I have 1 box that is accessible from the internet, the centos box.
I want to see all my pis videos through that one interface, motioneye.
I thought I could have the one motioneye centos that had multiple motion pis video streams on it.
I though that through the centos motioneye I could alter settings on the moiton pis if I wanted to.

jcherndon, 2016-06-12:
OK, I got things working though I'm not sure how correct it is. What would be the correct layout for what I mentioned above. 1 server hosting a web page for multiple pi cameras?

ccrisan, 2016-06-12:
@jcherndon just install [motionEyeOS](https://github.com/ccrisan/motioneyeos) or motionEye + Raspbian on your PIs and use your existing central motionEye from the centox box to control them.

Oxize, 2016-06-12:
CentOS MotionEye as webserver remoting the Pi cameras?

jcherndon, 2016-06-12:
okie doke, I'll close this, thanks!

closed 2016-06-29 by jcherndon

DonGrego617, 2017-05-26:
Hi, 
Love the app.  Fabulous.  But, I'm having a problem connecting MotionEye on Pi #1 to the Camera on MotionEye on Pi #2 on my local network.  I try to add remote motioneye camera on PI #2, I enter the "http://192.168.1.22" address, leave the password "admin", leave the password blank (i have no password on my MotionEye on Pi #2...)  but the error I receive is:  Failed to cennect to 192.168.1.22 port 80:  Connection refused.

I have a standard, out-of-the box Noobs installation of Raspian, with SSH running.  That's it.  There is nothing else I did to this box (ie: no other services, servers, firewall changes, etc.)  Is there something where I have to open the port?  Thanks in advance!!!
[image]