motioneye-project/motioneye #220: Image rasbian?

Oxize, 2016-06-12:
Does anyone have a Rasbian Jessie with MotionEye working on it? I tried to install it yesterday following crisan his steps, but couldnt get it working. 

Would be nice if there was some image with Rasbian + MotionEye preinstalled on it so i could place it on my Pi3B.

iSlide, 2016-06-13:
Hi
I've installed MotionEye on Raspbian Jessie on Pi3, and works fine.
Are you sure that you download the correct version?
First time I download wrong version?for Pi3?

Oxize, 2016-06-13:
Yes i had Jessie version 27-05-2016 which is the lastest build. 

https://downloads.raspberrypi.org/raspbian_latest

I am using an Raspberry Pi 3B. I can try again to install it and let you guys know were it got stuck.

Oxize, 2016-06-13:
When i add bcm2835-v4l2 to /etc/modules, it says at at startup this:

[image]

sgiannulli, 2016-06-13:
CSI camera: enable it via `sudo raspi-config` and append `bcm2835-v4l2` to your `/etc/modules` file, then reboot.
do you mean `/usr/local/share/motioneye/extra/motioneye.conf.sample` is missing?

Oxize, 2016-06-13:
Ye i had that with package from mrdave. I didnt run that today, and now i have the config sample.
Ive enabled the camera now also and tested it with modprobe. Camera is working.

Seems i can reach MotionEye now.

[image]

Thanks!

closed 2016-06-13 by Oxize