motioneye-project/motioneye #221: Custom hostname?

Fudge0952, 2016-06-13:
MotionEyeOS has this feature

ref: https://github.com/ccrisan/motioneyeos/issues/103

Just wondering if it's coming to the standalone?

ccrisan, 2016-06-13:
Nope. motionEye has nothing to do with the hostname of the system it runs on.

closed 2016-06-13 by ccrisan

Fudge0952, 2016-06-13:
I mean like being able to change the hostname of it? OS version has a Hostname: ____ field. Right now my pi box is named "pi2" but when I connect is there any way I can possibly have a more user friendly named like "Surveillance System" without compromising the hostname field for easy ssh'ing?

ccrisan, 2016-06-13:
Unfortunately no. The only thing you can do is to change the title of the window with ?title=sometitle

Fudge0952, 2016-06-13:
Ok no worries

ioquatix, 2017-09-10:
My system has multiple hostnames. How do I select which one is shown?

ccrisan, 2017-09-11:
@ioquatix you have posted on both motionEye and motionEyeOS issue trackers. Which one are you really using?

motionEye will display whatever `socket.gethostname()` returns. You could override the title if that's what's bothering you, using ?title=sometitle.

ioquatix, 2017-09-11:
I'm using both -I'm using one on a RPI for a camera and one on a arch linux box as a public server.