motioneye-project/motioneye #222: Cpu Load after reboot

iSlide, 2016-06-13:
Hi,
I've MotionEye installed with Raspbian.
I looked that when I start the Raspbian the cpu load is 0%, after the first access to MotionEye by browser,the cpu load is 23-25% and also when I logout of MotionEye the cpu load is stable to 25%. 
Now I'm not sure that the MotionEye automatic start with Raspbian, but only after first Login.
Is it correct this procedure? Is correct cpu load?
Many thanks for your support

ccrisan, 2016-06-13:
That sounds about right.

iSlide, 2016-06-14:
perfect, I'm very happy.
I've a question, is motioneye active also without first access by browser with cpu load 0%?

many thanks

ccrisan, 2016-06-14:
motionEye is "active" but streaming starts on demand. It will stop after 10 seconds of idling.

closed 2016-06-14 by ccrisan