motioneye-project/motioneye #224: show frame changes (motion detect)

iSlide, 2016-06-13:
Hi,
I've a little problem with motion detect.
I've 6 cameras, but only one has motion detect ON.
The problem is that when I turn on the option "show frame changes" on this camera (with motion detect) I see the frame on another camera with motion detect power off.
I tried several thing without success.

Can you help me?
Many thanks

ccrisan, 2016-07-03:
Can I see a screenshot of this problem manifesting? Can you please send me your config files (motion.conf and thread-*.conf), after having removed sensitive data from them?

iSlide, 2016-07-04:
Hi,
many thanks for you support!
Unfortunately I tried different solutions, but without results. I tried to delete camera and then add it again, I tried to delete all folder with videos ect.
The problem isn't only one camera but for all cameras. 
I have 7 cameras, if I active the motion detect and show frame for example on camera 2 I see the number of frames on the following camera (number 3)and the motion works on the following camera (number 3), and the data saved is of following camera (number 3). It's very strange!
If I active the motion detect on all camera all works fine.

ccrisan, 2016-07-04:
Please attach your config files, w/o sensitive data.

iSlide, 2016-07-04:
Hi,
I solved the problem! 
I changed this file: motion.conf in this folder etc/motioneye.
The original file was:
....
thread-1.conf
thread-4.conf
thread-6.conf
thread-5.conf
thread-2.conf
thread-3.conf
thread-9.conf
....

I changed the file in this mode:
....
thread-1.conf
thread-2.conf
thread-3.conf
thread-4.conf
thread-5.conf
thread-6.conf
thread-9.conf
....

in the same order of overview. Now works fine!
Why the orginal file was in disorder?

Many thanks

ccrisan, 2016-07-04:
Eh well, things are a bit more complex than they seem at first with motion. I'll try to take care of reordering these threads as soon as I manage to implement visual frame reordering in the UI.

iSlide, 2016-07-04:
many thanks for you support and for your fantastic Motioneye! Have a nice day,

closed 2016-07-04 by iSlide

ccrisan, 2016-10-26:
eef55b4c24a02c7840f95faaf43c52276749eafb should fix this problem for good.