motioneye-project/motioneye #225: motion recording not working on rpi3

eiri1234, 2016-06-13:
this is the 3rd installation i have tried to no avail, motion eye will not record anything. If anyone can help would much appreciate, Thank You in advance.

ccrisan, 2016-06-13:
Can you send me your config files (motion.conf, thread-*.conf)? Please make sure to remove passwords and other sensitive data from them. Also, what version of motionEye do you run?

eiri1234, 2016-06-13:
Thank you for answering  me so fast calin, I really appreciate your work so
much. For the rpi1 and 2 the settings for video and still images are by
default turned on but for rpi3 it is not and all I had to do is switch to
on for both. Thanks for your email and all your work.

ccrisan, 2016-06-14:
So the problem is gone then? Can we close the issue?

eiri1234, 2016-06-14:
Yes all done thanks.

closed 2016-06-14 by ccrisan