motioneye-project/motioneye #227: authentication

jfbdevs, 2016-06-14:
I just installed motioneye, but unfortunately I am unable to understand how authentication works.
I can access with admin - no password.
Where can I change this? I couldn't see it in the wiki.
Thanks!

ccrisan, 2016-06-15:
Use the admin password field.

closed 2016-06-15 by ccrisan

jfbdevs, 2016-06-15:
thanks ccrisan, sorry to be such a beginner... but can I ask you where do I find it the password field? I have been looking around the UI and found only how to enable authentication for streaming..

ccrisan, 2016-06-15:
It's the 7th field from the top in the settings menu.

jfbdevs, 2016-06-15:
ok I guess I have some problems to see the menu then... sorry I will check again and post if I have problems

ccrisan, 2016-06-15:
posting a screenshot could help.