motioneye-project/motioneye #229: [Feature] Option to manually start and stop recording?

Fudge0952, 2016-06-16:
The title says it all but is there any feature that allows the user to essentially press a record button on one or all webcams at any given time.

Say, for example, I don't have motion detected recording enabled but whilst viewing, I want to manually record a block of time; be it 5 minutes or 2 hours. Is there any support for this? 

I would assume it would just be a pop-up menu item like viewing photos and videos where I can simply press record to start recording and then stop to stop it.

Fudge0952, 2016-06-16:
Actually just noticed something very similar on the road map. Any idea on when you think the feature will be implemented?

ccrisan, 2016-06-16:
There are plans in this direction but no work has been done yet.

yanoschik, 2016-07-21:
I'm interested in this feature too... would appreciate if there were buttons to manually: A) take a picture, and B) start/stop recording. Thanks

dintid, 2017-07-14:
Any news on this? 
It's a bit of a bother (a large bit) having to change a setting and apply in order to manually record something and stop it again.
Thanks

Shaka-60hp, 2017-12-01:
I'm interested too, this post is rather old , Any news on this?

TopperBG, 2017-12-04:
Following

WaGi-Coding, 2018-09-26:
still not possible? :/

GrahamDB, 2018-12-05:
I am also interested in this feature.  Perhaps a command line script as a short term work around until this can be put in the web interface?

Palloquin, 2019-02-14:
+1 would like to see this feature!

crahlfs, 2019-02-18:
+1 me too. I need to send start / stop from with http request! Perfect for home automation

EJNash1, 2019-04-20:
+1 Any update on this. Is this project still ongoing as of Q2 2019?

jasaw, 2019-04-23:
As a workaround for now, you could start/stop recording by enabling motion detection, then sending a HTTP request to trigger motion detection.
`http://localhost:7999/1/config/set?emulate_motion=1` to force trigger motion detection.
`http://localhost:7999/1/config/set?emulate_motion=0` to disable forced motion detection.

ualex73, 2019-04-23:
The http request with eventstart and eventend should also work.

karthickshiva18, 2019-08-29:
> As a workaround for now, you could start/stop recording by enabling motion detection, then sending a HTTP request to trigger motion detection.
> `http://localhost:7999/1/config/set?emulate_motion=1` [...]

in which version this works...Please answer.

jasaw, 2019-08-29:
@karthickshiva18 Emulate motion has been in motion for a long time. Any version should work.

colinlgray, 2020-07-04:
@jasaw is there any more information how to send those commands via http request? I got a 404 error when sending both get and post requests. Screenshots attached from my terminal and motioneye.log

[image]

[image]

davebdb, 2020-07-05:
@colinlgray is motion running under port 80?  You need to specify the port
http://host:port/1/config/set?emulate_motion=1

here is my sample of my script for on and for off

```
$ cat light_on_1
#!/bin/sh
/usr/bin/curl "http://localhost:7999/1/config/set?emulate_motion=1"
```
```
$ cat light_off_1
#!/bin/sh
/usr/bin/curl "http://localhost:7999/1/config/set?emulate_motion=0"
```

port 7999 is the "command/control" for motion.  Your examples are going against port 80 so you must have some other webserver (or motioneye set) on that port.  

Let me know if that resolves your issue.

colinlgray, 2020-07-08:
Thanks for the clarification there. This was my bad, I am running motion via motioneyeos. It sets up a web interface on your local network and I was trying to send requests to that interface. I need to see how it exposes this interface

lcleve, 2020-07-29:
I need to record sleep, e.g. 8 hour sessions. A start/stop button would be very useful so that unskilled users could operate the soft ware. Even a toggle switch connected with one of the GPIO pins would be desirable. Currently, it looks like I will write a script to control rasspivid.

In any case the ease of installing and setting up MotionEyeOS is a welcome relief from software like BlueIris.

Many thanks, 

lcleve

jasaw, 2020-07-29:
@lcleve If you use a toggle switch, you should be able to use this script by just changing the GPIO pin: https://github.com/ccrisan/motioneyeos/issues/842#issuecomment-414375686

lcleve, 2020-07-29:
Thanks.
I'll look into it.

lcleve

rateur42, 2021-06-26:
+1 would like to see this feature!

WattageGuy, 2022-05-04:
+ 1

MichaIng, 2022-05-05:
A reasonable feature indeed, I would find it helpful for testing as well.

filip-strelec, 2023-12-25:
+

KodeMunkie, 2024-02-01:
+1 as it's a fairly easy feature to add to the ui

MichaIng, 2026-10-19:
The funny thing is that our action API supports `record_start` and `record_stop`, but nothing is done in the backend behind it. #2691 adds eventstart/eventend actions, which get just forwarded to the motion control API, as a very similar workaround to manually trigger a recording.

What is actually the practical difference between those and `emulate_motion=1`, if any? The `action` API does not have `webcontrol_parms` restrictions. As "action", it seems more explicit than changing a config value. But I do still wonder.

closed 2026-10-20 by MichaIng

MichaIng, 2026-10-30:
Action buttons for toggling motion detection and a motion event manually have been added with #3468. Manually triggering a recording only, without e.g. event notification, is difficult, since `motion` does not support such action, but is blocking the camera device, so a manual FFmpeg call our end cannot capture the video stream either. But starting/stopping a manual motion event, with motion-triggered video recording on, comes close.