motioneye-project/motioneye #231: Save movie files in mp4 format

10eputzen, 2016-06-17:
Is it possible to save the output files as in an .mp4 instead of .avi format?

ccrisan, 2016-07-03:
It's not done by motionEye. It's a motion thing and it's in the works.

ArmandH, 2016-07-28:
maybe idea to use mkv its faster and mkv supports H.264/AVC or choice between WebM, MOV, MKV or MPEG4.

ozett, 2016-08-17:
may use the file-save-hook ?

https://github.com/ccrisan/motioneye/issues/178#issuecomment-240327596

ccrisan, 2016-08-17:
This is actually already possible with the latest 0.33 version and a recent version of motion.

closed 2016-08-17 by ccrisan

ozett, 2016-08-17:
sorry to add here something to a closed thread,
but i want to prevent opening a non-issue. just a remark:

as i started to update motioneye +motion now on your suggestion,
after the long time i installed i forgot most of the details and started as a newbi.

with this point of view i followed the wiki (very good), but despite that i did not found the latest version of motion. the github repository front showed the latest commit from 2015, and after searching i found the release/pre-release page with recent development.

https://github.com/Motion-Project/motion/releases/tag/release-3.4.1

as i had issues with motion, i will try the latest with possible lots of improvements.
and the point here is: may you want to help others to with pointing from the very good and detailed wiki page for motion-install to the latest versions? could be irritating, what version to choose and later not to know, where the issues came from ... :smiling_imp: 
besides that minor remark  :+1:

ccrisan, 2016-08-17:
The motion daemon is currently in the process of rising from its own ashes and things change very quickly around it. As soon as things stabilize, I'll make sure to update the wiki docs accordingly.