motioneye-project/motioneye #235: Extermal motion trigger

Redferne, 2016-06-27:
Would like to request this feature for a http request and/or trigger file. So that we can get a doorbell to automatically capture the visitor.

ryanhoetger, 2016-10-22:
+1 I'm currently looking for a way to easily do this as a poor man's Ring

chrisf4lc0n, 2016-12-16:
OK guys bare with me on that one:
1. You need to have enabled Still Images.
2. Make Images File Name: Images/%T%d-%m-%y
3. Install ssmtp and configure it. (just Google it).
4. Install mpack. (Google it again).
5. To you doorbell activating script you will need to add the following:
```
d=`date +%d-%m-%y`
t=`date +%T
mpack -s "Somebody is at the door" /PATH/TO/YOUR/Images/$t$d.jpg your@email
```
That script assumes that there will be motion detected prior and during pressing the doorbell button!

chrisf4lc0n, 2016-12-25:
You can also refer to:
[https://github.com/ccrisan/motioneye/issues/94](https://github.com/ccrisan/motioneye/issues/94)

jorgekramer, 2018-02-03:
Returning to the initial issue:

I would like to trigger a snapshot via http-request (and then download the still-image to display it in my Home-Automation-System (FHEM)). Is there a way to do this?

ccrisan, 2018-02-05:
@jorgekramer you could simply use the Snapshot URL.

xxorde, 2019-08-26:
> @jorgekramer you could simply use the Snapshot URL.

@ccrisan I see its possible to get a still image via the Snapshot URL.
But like @Redferne I need to trigger motioneye to take a snapshot, like a virtual push on the snapshot button.
Than the image is saved by motioneye.

I have a doorbell (doorbird) and it can send an http(s) request whenever someone rings the bell or motion is detected.
It would be great if I could give it an API-URL that than triggers a snapshot or recording.

Is there such an URL already? Maybe we can use the manual snapshot function (action = 'record_start';).
If not, would it be a lot of work to implement?

salzrat, 2024-03-02:
I have the same situation. Actually I would really love to get the Doorbird motion sensor to trigger a video recording. The motioneye internal motion detection is just too hard to get working reliably, as it relies on image data alone. The Doorbird has a hardware motion detector so that would work much better...

JamBalaya56562, 2026-11-13:
motionEye 0.45.0b1 added this with #2691 and #3468: `POST /action/<camera id>/eventstart` starts a motion event, so a movie is recorded when movies are on, `eventend` ends it, and `snapshot` saves a picture when still images are on. A script logs in once and posts with the session cookie:
```
curl -c /tmp/me -d username=admin -d password=... http://<host>:8765/login/
curl -b /tmp/me -X POST http://<host>:8765/action/1/eventstart
```
A device that can only send a single plain request, like a doorbell, needs a small script in between for the login.

closed 2026-11-13 by JamBalaya56562