motioneye-project/motioneye #236: Notifications w/o pictures attached

sapcons, 2016-06-27:
I believe I am close to optimal settings for the two IP cameras I have attached to the system. 
However, sometimes I am receiving emails w/o pictures attached whereas pictures are available in the local folder. 
Also, in some cases out of a series of ten pictures only one is attached to the email - the last one, which does not show the 'trigger' of the motion event. How does motioneye select the picture?
Any advise?
BTW, thanks for all the great work!

ccrisan, 2016-06-29:
https://github.com/ccrisan/motioneyeos/wiki/FAQ#what-should-i-do-to-have-pictures-attached-to-motion-notification-emails

Marijn0, 2026-10-27:
Closing this old issue as per the comment above.

closed 2026-10-27 by Marijn0