motioneye-project/motioneye #237: Cleanup issue with multiple cameras

paolodemo, 2016-06-28:
Motioneye doesn't remove the old videos and picture.
In my configuration I have 3 cameras.
The first has not movie or image recording activated.
The other two have movie recording and picture recording with the preserve time of one week.

I think the problem is in the file [mediafiles.py, function cleanup_media](https://github.com/ccrisan/motioneye/blob/master/motioneye/mediafiles.py#L164).
The function do a loop over all the cameras, but when it checks for the [preserve_media](https://github.com/ccrisan/motioneye/blob/master/motioneye/mediafiles.py#L179) and it finds to preserve forever, then it returns. I think it is wrong because the other cameras are not processed.
The 'return' command should be replaced by a 'continue' command.

Anyway I want to thanks for your very good work!

ccrisan, 2016-07-01:
You are right! Thanks for spotting and reporting this bug.

ccrisan, 2016-07-01:
Fixed with 65f75426a227599ad7eb5ef053f695b16e8cfd5c.

closed 2016-07-01 by ccrisan