motioneye-project/motioneye #238: enable WiFi, motion not detected?

cyked82, 2016-06-28:
Let me know what logs you'd like to view..

I have a RPi 2 with the 5mp cam and it ran fine for a number of days.  It seems when I enable wifi (5g won't connect) the settings all seem fine and i can view the stream without issue, log in, etc. but motion is not detected and no motion capture images are saved.

Has this been an issue before?

ccrisan, 2016-07-01:
Normally there's no connection between the motion detection and the network connection. Check your motion detection settings and make sure you have disabled working schedule.

cyked82, 2016-07-01:
Thanks for replying.  I swear I'm not crazy and the initial setup was ethernet, motion sense on with motion notification emails.  I feel like enabling wifi flipped the flag on motion detection.  At any rate, motion was disabled and it works fine (of course) when motion detection is on.

closed 2016-07-01 by cyked82