motioneye-project/motioneye #239: Gmail refuse to send mail from motioneye

cpagura, 2016-06-30:
Motioneye 032 on Raspberry PI3 and raspicam NOIR v3.

All is ok, but It seems that now gmail server refuses to accept mail from "low security standard" app. As the matter of fact may be that meyectl when send e-mail is not "trusted" by gmail servers. 
Moreover, the motioneye interface write in clear all the parameters from web interface, pswd included, in "thread-1.conf", not a good policy, imho ...  ;-).

I will try another way to send e-mail notification ...

cpagura, 2016-06-30:
I found the solution: increase the tolerance of gmail account! see:
https://support.google.com/accounts/answer/6010255?hl=en now all is working! ... even if not 100% safely :-)

closed 2016-06-30 by cpagura

ccrisan, 2016-07-01:
@cpagura I agree that reducing the security level of your Google account is not desirable. However there is another way where you actually increase the security level while enabling separate apps to access your account the old way: https://support.google.com/accounts/answer/185833. In a nutshell, you enable two-steps verification for your Google account and you create separate passwords for each app you want to give access to your account. You'll then need to create a password for motionEye.

pdesimonee, 2021-09-05:
@cpagura: I cannot find this thread-1.conf file in the email I receive. Help

starbasessd, 2021-09-05:
@pdesimonee Open a new issue with your specifics. 
This issue was closed 5 years ago, and MANY things ( including motion, motionEye, and Google )have changed since then. I would be more than happy to help, but need YOUR CURRENT specifics.

pdesimonee, 2021-09-05:
ok. Why, when I disable the email notification the configuration of the email get lost?. Sometime I want the email [when I am on leave] sometimes not. But the configuration should not be lost every time I switch off the notification. How to go around it?

starbasessd, 2021-09-05:
@pdesimonee ignored for not following directions. This issue is closed and now locked.