motioneye-project/motioneye #240: Monitoring commands

Redferne, 2016-07-02:
Hi, 

Please help getting the monitoring command working, followed the example below but I get nothing overlayed on the stream. No usable information in the debuglog. 
https://github.com/ccrisan/motioneye/wiki/Monitoring-Commands

BR, 
Rickard

ccrisan, 2016-07-03:
1. What version?
2. Where have you placed the file?
3. What name did you give the file?
4. Did you make it executable?
5. If you run it manually from the command line, does it return immediately and output the expected text? If yes, which is the text?

maglub, 2016-07-03:
You will get the output of the monitoring command in the "border" of the camera, when you click on the camera in question.

[image]

Redferne, 2016-07-03:
Ah, now I see. It works here aswell, but I was expecting it to be "burnt
into the stream/screenshot".
Is it possible to add this functionality? Cameraname and Timestamp is
currently beeing "burnt into the image".
That would be a very powerful option, we can have a small table on the
topleft with all kind of information :-)

BR,
Rickard

ccrisan, 2016-07-03:
Nope, that's not possible as it would require changing the source code of motion.

closed 2016-07-03 by ccrisan

maglub, 2016-07-04:
A workaround would be to use the MotionHttpApi, sending the text you would like to burn in per "curl".

I did this:

* On the motioneye, find the "webcontrol_port" which should be 7999

```
[root@meye-77c702cd etc]# grep control *
motion.conf:webcontrol_localhost on
motion.conf:webcontrol_html_output on
motion.conf:webcontrol_port 7999
```

* Use "curl" to send a new text to the text you want:

```
[root@meye-77c702cd etc]# curl http://localhost:7999/2/config/set?text_left="my%20text"
```

Before:

[image]

After:

[image]

ccrisan, 2016-07-04:
@maglub actually that is a great idea indeed. I never thought of that. Thanks for sharing it with us.