motioneye-project/motioneye #241: motion-mrdave-raspbian compiled against libraries not found in Raspbian

maglub, 2016-07-04:
Hi,

The instruction in the raspbian installation tells us to install libavutil54, but the Mr Dave motion binary you offer is compiled against libavutil55

```
pi@me-r001fry:~/motion/motion $ cat /etc/os-release 
PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
NAME="Raspbian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

pi@me-r001fry:~ $ ldd /usr/local/bin/motion 
	linux-vdso.so.1 (0x7eff0000)
	/usr/lib/arm-linux-gnueabihf/libarmmem.so (0x76f90000)
	libm.so.6 => /lib/arm-linux-gnueabihf/libm.so.6 (0x76f0d000)
	libpthread.so.0 => /lib/arm-linux-gnueabihf/libpthread.so.0 (0x76ee5000)
	libjpeg.so.62 => /usr/lib/arm-linux-gnueabihf/libjpeg.so.62 (0x76e90000)
	libavutil.so.55 => not found
	libavformat.so.57 => not found
	libavcodec.so.57 => not found
	libswscale.so.4 => not found
	libc.so.6 => /lib/arm-linux-gnueabihf/libc.so.6 (0x76d4e000)
	/lib/ld-linux-armhf.so.3 (0x54b6e000)
pi@me-r001fry:~ $ ls -la /usr/local/bin/motion 
-rwxr-xr-x 1 root staff 224016 Jul  4 14:25 /usr/local/bin/motion
pi@me-r001fry:~ $ md5sum /usr/local/bin/motion 
7a01eae7d0ceb80483a7fe99efb89cec  /usr/local/bin/motion
```

Workaround, compile motion from Mr Dave fork, as described in https://github.com/ccrisan/motioneye/wiki/Compiling-Motion, but I just can't figure out the dependencies. The closest I get give me the following errors:

```
pi@me-r001fry:~/motion/motion $ ./configure --prefix=/usr --with-ffmpeg=/usr/lib/arm-linux-gnueabihf  --with-ffmpeg-headers=/usr --without-pgsql --without-sdl --without-sqlite3 --without-mysql
configure: WARNING: unrecognized options: --with-ffmpeg-headers
...
make
...
gcc  -o motion motion.o logger.o conf.o draw.o jpegutils.o vloopback_motion.o video.o video2.o video_common.o netcam.o netcam_ftp.o netcam_jpeg.o netcam_wget.o track.o alg.o event.o picture.o rotate.o webhttpd.o stream.o md5.o netcam_rtsp.o ffmpeg.o  -lm  -lpthread -ljpeg  -lavformat -lavcodec -ldc1394 -lxvidcore -lx264 -lvorbisenc -lvorbis -ltheoraenc -ltheoradec -logg -lopencore-amrwb -lopencore-amrnb -lmp3lame -lgsm -lz -pthread -lswresample -lswscale -L/usr/local/lib  -lavutil -lm  
/usr/bin/ld: cannot find -lxvidcore
/usr/bin/ld: cannot find -lx264
/usr/bin/ld: cannot find -lvorbisenc
/usr/bin/ld: cannot find -lvorbis
/usr/bin/ld: cannot find -ltheoraenc
/usr/bin/ld: cannot find -ltheoradec
/usr/bin/ld: cannot find -logg
/usr/bin/ld: cannot find -lmp3lame
/usr/bin/ld: cannot find -lgsm
collect2: error: ld returned 1 exit status
Makefile:82: recipe for target 'motion' failed
make: *** [motion] Error 1
```

ccrisan, 2016-07-04:
You're right I have edited the file yesterday and I forgot to change some parts of it. The idea now is to use ffmpeg-3.1.1 (which comes with libavwhatever.55) and to always use Mr Dave's version of motion, preventing users to go the easy way and install the old motion from official repos (which has only led to tons of problems so far).

I have now updated the wiki page and it no longer suggest to reinstall those .54 libs individually, but rather insists that you install the new ffmpeg. Let me know if you can make it work now.

maglub, 2016-07-04:
libx264.so.142 seems to be missing

```
pi@raspberrypi:~ $ ldd /usr/local/bin/motion 
	linux-vdso.so.1 (0x7ef4d000)
	/usr/lib/arm-linux-gnueabihf/libarmmem.so (0x76fc8000)
	libm.so.6 => /lib/arm-linux-gnueabihf/libm.so.6 (0x76f47000)
	libpthread.so.0 => /lib/arm-linux-gnueabihf/libpthread.so.0 (0x76f1f000)
	libjpeg.so.62 => /usr/lib/arm-linux-gnueabihf/libjpeg.so.62 (0x76eca000)
	libavutil.so.55 => /usr/lib/libavutil.so.55 (0x76e41000)
	libavformat.so.57 => /usr/lib/libavformat.so.57 (0x76c65000)
	libavcodec.so.57 => /usr/lib/libavcodec.so.57 (0x75b55000)
	libswscale.so.4 => /usr/lib/libswscale.so.4 (0x75ae2000)
	libc.so.6 => /lib/arm-linux-gnueabihf/libc.so.6 (0x759a1000)
	/lib/ld-linux-armhf.so.3 (0x54b7a000)
	libz.so.1 => /lib/arm-linux-gnueabihf/libz.so.1 (0x7597a000)
	libswresample.so.2 => /usr/lib/libswresample.so.2 (0x75959000)
	libx264.so.142 => not found
```

maglub, 2016-07-04:
Workaround, install libx264-142

```
sudo apt-get install libx264-142
```

Now your supplied "motion" works:

```
pi@raspberrypi:~/src/x264 $ motion -h
motion Version Unofficial-Git-UNKNOWN, Copyright 2000-2005 Jeroen Vreeken/Folkert van Heusden/Kenneth Lavrsen
```

ccrisan, 2016-07-04:
I have added libx264-142 to the dependencies list in the wiki page. Thanks.

closed 2016-07-04 by ccrisan