motioneye-project/motioneye #242: how to obtain the filename of last picture saved in /var/lib/motioneye?

cpagura, 2016-07-05:
I'm trying to send a DM tweet using the tweepi library and a little .py program as a command to be run on motion notification.
It works, but if I want to add the last picture saved by motioneye on /var/lib/motioneye, how to add the file name as a parameter for tweet_foto.py?
example of command:
python /home/pi/tweet_foto.py /var/lib/motioneye/2016-07-05/08-58-35.jpg

code of tweet_foto.py:

import tweepy
import sys
import datetime
import time

# chiavi di accesso per Twitter
consumer_key = '8umfMKxxxxxxxxxxxxxxxxxxxxxxxx'
consumer_secret = 'F7DCa7WDYG9JmgSBxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx'
access_token = '712950117893029888-krrcxxxxxxxxxxxxxxxxxxxxxxxxxx'
access_token_secret = 'fi9wvjvFya4YBB4SLk0Cxxxxxxxxxxxxxxxxxxxxxxxxxxxxx'

# autenticazione su twitter
auth = tweepy.OAuthHandler(consumer_key, consumer_secret)
auth.set_access_token(access_token, access_token_secret)
api = tweepy.API(auth)

# genero la stringa con data/ora attuale
ts = time.time()
st = datetime.datetime.fromtimestamp(ts).strftime('%Y-%m-%d %H:%M:%S')

# utente che deve essere notificato
twitter_user = "@xxxme"

# controllo l'esistenza di due argomenti nel lancio del file py
if len(sys.argv) >=2:
   tweet_image = sys.argv[1] #here the argument containing the complete file name from motioneye
   tweet_text = twitter_user + " Movimento rilevato! (" + st + ")"
else:
   tweet_text = twitter_user + " Qualcosa non va nello script del tweet..."
#tweet_text = twitter_user + " prova di twitting da raspberry " + st
# mando il tweet
api.update_with_media (tweet_image, status=tweet_text)

print tweet_text
... sorry for the comment in italian ...

cpagura, 2016-07-05:
the code is from Mr. Fabrizio Tucci blog: http://www.iltucci.com/blog/2014/07/08/videosorveglianza-con-raspberry-pi-parte-prima/

ccrisan, 2016-07-05:
The last picture will always be symlinked to by /var/lib/motioneye/lastsnap.jpg.

closed 2016-07-05 by ccrisan

cpagura, 2016-07-06:
Ok! thank you Calin ... I tried but I got an error "No such file or directory" ... as the matter of fact with "locate" command I cannot find any trace of link file ... could be a version problem? or the creation of link is disabled ?

ccrisan, 2016-07-06:
@cpagura please describe your setup a bit: what hardware, what OS, what firmware version?

cpagura, 2016-07-06:
HW: Raspberry Pi 3 model  V 1.2; Raspicam Pi NoIR V.2
FW: Jun 20 2016 18:18:51
Copyright (c) 2012 Broadcom
version 460b39c674ef2ac89360e15331f97a2cb1388a44 (clean) (release)
OS:Raspbian GNU/Linux 8 (jessie)
SW: motioneye 0.32

cpagura, 2016-07-06:
... upgraded just now to motioneye-0.32.2.

ccrisan, 2016-07-06:
@cpagura What is your Root Directory set to?

cpagura, 2016-07-06:
Entering Raspberry with a SSH session I normally see the directory  /home/pi/

ccrisan, 2016-07-06:
That's correct. However I was asking about the Root Directory setting in the File Storage section of your motionEye's settings panel.

cpagura, 2016-07-06:
Ops .. :-) sorry ... The root Directory in file storage setting is /var/lib/motioneye. In such directory now there are the file .keep and the subdir 2016-07-05 containing all the images of yesterday tests.

ccrisan, 2016-07-06:
What does `find /var/lib/motioneye -name lastsnap.jpg` say?

cpagura, 2016-07-06:
nothing ... :-(

ccrisan, 2016-07-06:
Have you enabled Still Images at all in your camera settings?

cpagura, 2016-07-06:
Oh yess! leaving the default for file name: %Y-%m-%d/%H-%M-%S and Image quality and setting Capture Mode for Motion triggered and preserve Pictures for one day

ccrisan, 2016-07-06:
Then it's probably deleted by the preserve pictures option. I recommend using the *File Storage > Run A Command* option to call your command each time a new picture is created. You could simply use "cp %f /tmp/lastfile.jpg" to copy each picture to some specific location, just as the lastsnap.jpg would work.

cpagura, 2016-07-06:
Ok ... thank you very much ... Let me try!

cpagura, 2016-07-06:
Ah! Thou! Gigantic Guru! Sir Crisan Master of Keys of Recondite Secrets Of Motion Castle! :-) Thank you very very much!

... it works now !

rainman-xx, 2018-07-31:
@ccrisan , How about the last recorded video?
I have Pi3 with two wireless camera setup
```

motionEye Version | 0.39.2
Motion Version | 4.1.1
OS Version | Raspbian 9.4

$ uname -a
Linux raspberrypi 4.14.50-v7+ #1122 SMP Tue Jun 19 12:26:26 BST 2018 armv7l GNU/Linux
cat /proc/version
Linux version 4.14.50-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1122 SMP Tue Jun 19 12:26:26 BST 2018
```

I plan to use ffmpeg to upload (through rtsp streaming) my last motion-triggered video to my youtube private channel. I just need to know where to get it accurately. I will be calling ffmpeg within "Run An End Command"