motioneye-project/motioneye #244: Network Share Option not offered

vtrack, 2016-07-06:
Hi,
I just installed the latest version of motioneye on raspbian, following the instructions in the wiki.
0.32.2
Linux raspberrypi 4.4.14+ #896 Sat Jul 2 14:16:46 BST 2016 armv6l GNU/Linux

compiled latest trunk of motion
ffmpeg from your pre-compiled url.

I am not seeing the option to save files to a network share in the dropdown. in File Storage/Storage Device. I only see "Custom Path" or SL16G with a path to a local device.
See attached configs.

- pi@raspberrypi:/etc/motioneye $ cat thread-1.conf
- # @webcam_resolution 100
- # @upload_subfolders on
- # @upload_server
- # @enabled on
- # @network_server
- # @upload_username
- # @motion_detection on
- # @upload_port
- # @upload_location
- # @preserve_movies 0
- # @network_username
- # @upload_movie on
- # @id 1
- # @upload_password
- # @upload_method post
- # @upload_picture on
- # @working_schedule_type outside
- # @webcam_server_resize off
- # @network_password
- # @upload_service gdrive
- # @name piCamera
- # @preserve_pictures 0
- # @storage_device custom-path
- # @network_share_name
- # @upload_enabled off
- # @working_schedule

> 
> ffmpeg_output_movies on
> height 240
> stream_quality 85
> threshold 1999
> quality 85
> noise_level 31
> pre_capture 1
> noise_tune on
> stream_maxrate 5
> output_pictures off
> hue 0
> saturation 0
> stream_localhost off
> ffmpeg_variable_bitrate 10
> ffmpeg_video_codec msmpeg4
> text_changes off
> movie_filename %Y-%m-%d/%H-%M-%S
> auto_brightness off
> stream_port 8081
> rotate 0
> brightness 0
> lightswitch 0
> framerate 2
> emulate_motion off
> snapshot_filename %Y-%m-%d/%H-%M-%S
> snapshot_interval 3
> stream_auth_method 0
> stream_motion off
> target_dir /var/lib/motioneye
> text_double off
> post_capture 1
> stream_authentication user:***********************
> on_picture_save /usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" picture_save %t %f
> on_movie_end /usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" movie_end %t %f
> text_left piCamera
> picture_filename
> locate_motion_style redbox
> locate_motion_mode off
> contrast 0
> videodevice /dev/video0
> max_movie_time 0
> on_event_end /usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" stop %t
> text_right %Y-%m-%d\n%T
> on_event_start /usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" start %t
> event_gap 30
> minimum_motion_frames 20
> width 320
> 
> 
>

-  pi@raspberrypi:/etc/motioneye $ cat motioneye.conf
- > # the base URL path at which motionEye lives
- > # change this if you run motionEye behind a reverse proxy (e.g. nginx),
- > # and you want to make motionEye accessible at a specific base path (e.g. /cams)
- > #base_path /cams
- > 
- > # path to the configuration directory (must be writable by motionEye)
- > conf_path /etc/motioneye
- > 
- > # path to the directory where pid files go (must be writable by motionEye)
- > run_path /var/run
- > 
- > # path to the directory where log files go (must be writable by motionEye)
- > log_path /var/log
- > 
- > # default output path for media files (must be writable by motionEye)
- > media_path /var/lib/motioneye
- > 
- > # path to the motion binary to use (automatically detected if commented)
- > #motion_binary /usr/bin/motion
- > 
- > # the log level (use quiet, error, warning, info or debug)
- > log_level info
- > 
- > # the IP address to listen on
- > # (0.0.0.0 for all interfaces, 127.0.0.1 for localhost)
- > listen 0.0.0.0
- > 
- > # the TCP port to listen on
- > port 8765
- > 
- > # whether motion HTTP control interface listens on
- > # localhost or on all interfaces
- > motion_control_localhost true
- > 
- > # the TCP port that motion HTTP control interface listens on
- > motion_control_port 7999
- > 
- > # interval in seconds at which motionEye checks the SMB mounts
- > mount_check_interval 300
- > 
- > # interval in seconds at which motionEye checks if motion is running
- > motion_check_interval 10
- > 
- > # interval in seconds at which the janitor is called
- > # to remove old pictures and movies
- > cleanup_interval 43200
- > 
- > # timeout in seconds to wait for response from a remote motionEye server
- > remote_request_timeout 10
- > 
- > # timeout in seconds to wait for mjpg data from the motion daemon
- > mjpg_client_timeout 10
- > 
- > # timeout in seconds after which an idle mjpg client is removed
- > # (set to 0 to disable)
- > mjpg_client_idle_timeout 10
- > 
- > # enable SMB shares (requires motionEye to run as root)
- > smb_shares false
- > 
- > # the directory where the SMB mount points will be created
- > smb_mount_root /media
- > 
- > # path to the wpa_supplicant.conf file
- > # (enable this to configure wifi settings from the UI)
- > #wpa_supplicant_conf /etc/wpa_supplicant.conf
- > 
- > # path to the localtime file
- > # (enable this to configure the system time zone from the UI)
- > #local_time_file /etc/localtime
- > 
- > # enables shutdown and rebooting after changing system settings
- > # (such as wifi settings or time zone)
- > enable_reboot false
- > 
- > # timeout in seconds to use when talking to the SMTP server
- > smtp_timeout 60
- > 
- > # timeout in seconds to wait for media files list
- > list_media_timeout 120
- > 
- > # timeout in seconds to wait for media files list, when sending emails
- > list_media_timeout_email 10
- > 
- > # timeout in seconds to wait for zip file creation
- > zip_timeout 500
- > 
- > # timeout in seconds to wait for timelapse creation
- > timelapse_timeout 500
- > 
- > # enable adding and removing cameras from UI
- > add_remove_cameras true
- >

vtrack, 2016-07-06:
still researching this. Is this something with the motioneye feature-set? I am coming from MotionPie and was hoping to have more or less the same features. If not, I would use motionos, but I need the ability to add repos and use the gpio.

ccrisan, 2016-07-06:
Set [smb_shares](https://github.com/ccrisan/motioneye/wiki/Configuration-File#smb_shares) setting (in motioneye.conf) to `true`.

closed 2016-07-07 by vtrack

stjowa-gn, 2016-08-05:
When I set to true the service will not start.

jasonhpchu, 2017-08-15:
After tinkering for a while, I've found the issue that none of the logs told me.
running this directly, it gave me the log on the stream, which told me the problem:
 /usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf

it'll say CRITICAL: please install cifs-utils
now, going to edit the server script to have it log to file

MarcoMirisola, 2018-01-28:
Hi, I think that @jasonhpchu solutions should be added in the installation guide!

Frank-Ov, 2020-01-09:
Hello

i have a similar problem.
3 cameras attached on MotionEye ( Debian ) - two Raspi  ( "0" and PI3 ) and one Instar IN-3011cam. 
Everything works fine - thank you for the great work.
Just missing the option "File Storage - Network Share " on the Instar cam. Its working on both PI´s perfectly. Local storage is available

 and working cam is working pretty well.
Is it because of the Instar  cam ?? 
[image]
Thank you for help !
Grüße
Frank

fgtome, 2025-01-12:
Hi guys, so I am having the same problem here.

I'm running Motion Eye over Debian 12 in arm64 (on my Raspberry Pi 4). Because I did not have the network share option on any camera, I changed the smb_shares to true, and after rebooting, I lost access to the GUI.

Can someone help me fix this, please? I still have access to it via SSH, though.

Both cameras are built with Raspberry Pi Zero W running motioneyeOS in "Fast Camera Mode".

Thank you all!