motioneye-project/motioneye #245: Feature request

ChriZathens, 2016-07-07:
Hello!
I would like to request for a possibility to edit the camera via the interface.
My old foscam 8910w died last week and I purchased a dlink dcs-5010l
I wanted all the same settings, except from the url to the video of the camera (the old one was using http://camera_ip/videostream.cgi and the new one is using http://camera_ip/video.cgi.)
So all i had to do was delete 6 letters and I should be good to go. Instead I had to create a second camera and go back and forth between the two cameras to apply the same settings to the new one as they were in the old one and finally delete the old. (I initially tried editing the config backup and restore it but did not work...)
Alternatively, if this feature is not possible, the possibility to backup only the motioneye settings from a specific camera and being able to restore them to any camera, would be a nice feature!
What do you think???

mce35, 2016-11-03:
Hello,
For your specific case you could have edited the thread-X.conf (in /etc/motioneye) by hand to change netcam_url. I don't know why modifying netcam_url is not possible in the Web UI.

But as you suggest applying all settings from one camera to another would be great. I have 6 cameras all with the same settings, I have edited all thread-X.conf by hand.

Having an easy way to duplicate a camera or some sort of template/camera profile would be easier.

ChriZathens, 2016-11-03:
Yeap after doing all that I realized I could have edited thread file, but still, the possibility to edit the url from the webui and/or backup /restore camera settings, would be a good addition, I believe

ccrisan, 2019-06-21:
This is a feature whose implementation is a lot more difficult than it seems and whose benefits are minimal. So I'm going to pass.

closed 2019-06-21 by ccrisan