motioneye-project/motioneye #246: Any chance of adding motion boxes or areas?

fundean, 2016-07-07:


ccrisan, 2016-07-08:
It's in the works.

fundean, 2016-07-08:
That is absolutely awesome... Like a kid on Christmas Eve!! Thank you

kurazli, 2016-07-12:
using the area_detect feature of motion does not work i guess.
really looking forward for this feature!

fundean, 2016-07-12:
I know your busy ccrisan, but if you get a chance check out Logitech security's software?!
Prob the best I have seen at the boxes... You select as many areas (square boxes infinitely sizable) for the areas you want motion active.
Ps I have the software if that helps...

quantum8, 2016-08-11:
Looking forward to this being implemented, keep up the great work!

ccrisan, 2016-08-11:
Still in the works.

SuperSharpShooter, 2016-08-15:
Very much hanging out for this. Where I live often has fast moving clouds, and a tree that blows in the wind, and this gives me 80% false positives and this makes it pretty much unusable.

sempervivo, 2016-08-31:
Hi Calin,
Detecting specific areas/zones would bring to the frontend much value. Please tell us if this feature will be ready anytime soon. 

Thank you,
George Dadulescu

ccrisan, 2016-08-31:
It won't be ready any time soon. In the meantime, you could always use the extra motion options to specify a mask file of your own.

jeff-h, 2016-08-31:
For those wanting to use a mask, there are a couple of other issues describing this:

* https://github.com/ccrisan/motioneye/issues/50
* https://github.com/ccrisan/motioneye/issues/47

Essentially you need to:

* create an image of the exact same pixel dimensions as your camera (be wary, if you have several cameras they may well produce images with different px dimensions). This image should be black-and-white (or greyscale). White means 100% sensitive, black means %100 ignore, and grey proportionally between those.
* then convert that image to `.pgm` format (there are several ways to do this, but I've used https://convertio.co/pgm-converter/)
* place this `pgm` file at a path that MotionEye can reach it, and enter the following into "Extra Motion Options":

    `mask_file /path/to/mymaskfile.pgm`

quantum8, 2016-09-01:
Perfect. Can we get this information added to the wiki instead of buried in multiple issue threads?

jeff-h, 2016-09-01:
You're right :) Doing it now.

jeff-h, 2016-09-01:
This is now in the wiki at https://github.com/ccrisan/motioneye/wiki/Ignoring-areas-of-the-image-using-masks

closed 2016-09-01 by jeff-h