motioneye-project/motioneye #247: Adding remote motioneye camera

KrisJoyce, 2016-07-08:
Thanks for motioneye I love it.

My issue is when I try to add a second camera in motioneye, in the camera drop down menu it says "no camera".
I have added one previously and worked, but if I try to add the first one again it shows "no camera" as well. Both camera are set for Fast Network camera.

Thanks

ccrisan, 2016-07-08:
You can have only one Fast Network Camera per motionEyeOS unit. There are however two things that are unclear to me:

* Fast Network Camera is available only with motionEyeOS, not with a plain motionEye install; are you actually running motionEyeOS, even though you have opened the issue for motionEye?
* When you have enabled the Fast Network Camera mode, the camera dropdown menu should no longer be displayed. Can I see a screenshot of your camera dropdown menu as well as of the Fast Network Camera setting?

KrisJoyce, 2016-07-08:
I have two raspberry Pi running motionEyeOS with 1 CSI camera on each. With a PC running MotionEye to do the motion detection and recording.

[image]
[image]
[image]

ccrisan, 2016-07-08:
If you're trying to add Fast Network Cameras to a motionEye server then you'll have to add them as... well... Network Cameras. They are not managed as motionEye cameras.

KrisJoyce, 2016-07-08:
When I try that I get "not a supported network camera".

ccrisan, 2016-07-08:
Most likely you tried to add the web interface (port 80). Use port 8081 or whatever streaming port you have configured for your Fast Network Camera.

KrisJoyce, 2016-07-08:
You help worked but now I get "An error occurred. Refreshing is recommended." I did this several time and it does not add the camera.

ccrisan, 2016-07-08:
Please send me your motioneye.log file. You can obtain it by running `journalctl -u motioneye > motioneye.log`.

KrisJoyce, 2016-07-08:
I did a quick check and found out my storage was full. I emptied it and was able to and the 2nd camera.

Thank you very much for all you help.

closed 2016-07-08 by KrisJoyce

ccrisan, 2016-07-08:
Happy to help.