motioneye-project/motioneye #248: google drive error

kuraz758, 2016-07-11:
one camera uploads fine, the other shows this in the log file:

2016-07-11 11:31:24: [motioneye]    ERROR: failed to upload file "/mnt/proworx/motioneye/2016-07-11/Kueche-11-29-43.avi" with service gdrive: 'NoneType' object has no attribute 'endswith'
2016-07-11 11:49:32: [motioneye]    ERROR: failed to upload file "/mnt/proworx/motioneye/2016-07-11/Kueche-11-48-09.avi" with service gdrive: 'NoneType' object has no attribute 'endswith'
2016-07-11 12:04:04: [motioneye]    ERROR: failed to upload file "/mnt/proworx/motioneye/2016-07-11/Kueche-12-00-57.avi" with service gdrive: 'NoneType' object has no attribute 'endswith'

i already tried a new authentication key.

ccrisan, 2016-07-11:
Can you set the log level to DEBUG in your motioneye.conf and then send me the entire log file?

kuraz758, 2016-07-11:
[motioneye.log.txt](https://github.com/ccrisan/motioneye/files/357292/motioneye.log.txt)

ccrisan, 2016-07-11:
You haven't set the log level to DEBUG. Or you have set it but never restarted motionEye.

kuraz758, 2016-07-11:
sorry, i blindly followed your instructions :)

[motioneye.log (2).txt](https://github.com/ccrisan/motioneye/files/357343/motioneye.log.2.txt)

kuraz758, 2016-07-11:
after also rebooting it seems to work fine again.

closed 2016-07-11 by kuraz758