motioneye-project/motioneye #249: precompiled motion for RPi in wiki instructions doesn't work with latest Raspbian

rbnpi, 2016-07-11:
The precompiled motion binary for RPi in the wiki instructions doesn't appear to work with the latest raspbian 2016-05-27 image with 4.4 kernel.
Is there an alternative that can be used?

motion -h gives response illegal  instruction

ccrisan, 2016-07-11:
How did you install ffmpeg and what version do you currently have?

rbnpi, 2016-07-11:
I installed the version in the wiki file.
```
wget https://github.com/ccrisan/motioneye/wiki/precompiled/ffmpeg_3.1.1-1_armhf.deb
dpkg -i ffmpeg_3.1.1-1_armhf.deb
```
then I installed motion from
```wget https://github.com/ccrisan/motioneye/wiki/precompiled/motion-mrdave-raspbian -O /usr/local/bin/motion
chmod +x /usr/local/bin/motion
```

In fact for the moment I have switched to use motioneyeos running successfully ion a PiZero with a wifi dongle, so this is not too urgent now, but I would like to get it going if possibile.

ccrisan, 2016-07-11:
Well I would also like to make sure this is actually working as many users prefer Raspbian + the precompiled stuff to motionEyeOS. So is your previous issue reproducible? Meaning that you download the latest Raspbian, you do the apt-get update + upgrade crap and you install the indicated pieces of software and then boom, Illegal Instruction?

What RPi board version is this, btw?

rbnpi, 2016-07-12:
Yes repeatable. Done it twice on new installs of 2016-05-27 image. apt-get update and apt-get upgrade then follow wiki page install. Running on a pizero with wireless config added and usb dongle also tried on model b+

ccrisan, 2016-07-12:
I'll try it myself again. Some new updates might have messed up things, again.

pleitao, 2016-07-16:
Hi all - just signed in to let you know I am facing the same problem with the new version of raspbian on my end. I have a rpi model B

ccrisan, 2016-07-16:
Yeap. It's fully reproducible and it only happens on RPi v1. I'll make sure to fix it asap.

ccrisan, 2016-07-16:
@rbnpi, @pleitao can you please try to redownload the precompiled motion version and try again? I have compiled it for RPi1 now which should work just fine for all RPi variants.

rbnpi, 2016-07-16:
Thanks Calin. All working fine now. I'll close this.

closed 2016-07-16 by rbnpi