motioneye-project/motioneye #250: Changing Admin username?

Fudge0952, 2016-07-12:
Don't know if it's by design, but is there any way of changing the administrator's username to access the system?

ccrisan, 2016-07-12:
It's by design.

closed 2016-07-12 by ccrisan

Bluscream, 2016-12-16:
Any news on this? i would like to choose a harder to guess admin name.

yed30, 2016-12-28:
I totally agree with @Bluscream. 
The way I found was to edit /etc/motioneye/motion.conf and change the below line like (in bold):
# @admin_username **myadminusername**
then restart MotionEye
I hope this help

ccrisan, 2016-12-28:
This works (and used to work this way) but when you'll try to log in using SSH, FTP or SMB, you'll see that your new admin user won't work. This is because it's difficult to update the username for all these services.

Bluscream, 2016-12-28:
@yed30 thank you very much <3

yed30, 2016-12-28:
Thanks @ccrisan for your answer. I just use MotionEye web frontend (which I find very handy ! :) ), behind a NGinx server, so I guess this shouldn't be a problem.

DelightedD0D, 2019-02-18:
Might be useful to have the ui show "admin" in the readonly field and remove the red exclamation icon indicating that its a "required field". Its super confusing as it stands as the UI tells you that you have to fill in this non-functional field.