motioneye-project/motioneye #251: de/activate motion detection by Script - not start/stop service

dfens-1373, 2016-07-16:
Is there a way just to de/activate motion detection by script (triggered by gpio) so that the service stays on and motioneye could be furthermore controlled by Web Interface (e.g. examine files or restart motion detection)?

And is it possibly to read from commandline/script if motion detection is enabled?

ozett, 2016-08-17:
you could do something like this to "motion " itself.
it has a "http-interface"/api

http://www.lavrsen.dk/foswiki/bin/view/Motion/RemoteControlHttp

ccrisan, 2016-08-17:
Nope, unfortunately that won't be possible because of the working schedule management which will continuously overwrite the setting. I really must do something about this.

ozett, 2016-08-17:
:+1: 
(i will update my motion + motioneye to the latest, to keep up on your improvments)
:smile:

pantchox, 2017-01-01:
@ccrisan hi! great job on motion! is there any workaround for this? .
i currently don't have schedule management enabled. thanks!

sapcons, 2017-07-01:
Hello, I am also very much interested in such a script. I am running CCU home automation system on my RPI. When I am at home I want to 'pause' alerting on motioneye. Recording can still go on, but I don't want to receive emails. The camera itself doesn't allow 'shutdown' or 'startup'. Thanks for all your effort!!!

SuperBFG7, 2017-11-03:
Hi, using the motion http-interface is possible as long as you don't use the woking schedule.
I'm using a small script to ping our phones/tablets and when any of them is online, turn off motion detection. Works pretty good for me.

https://github.com/SuperBFG7/scripts/blob/master/pi/surveillance.sh

The script also understands pause and start command, so you could use it in a crontab to setup your own schedule. The script remembers the online state and won't interfere (only changes in only presence trigger (de)activation of motion).

PS: remember that everytime you restart motion/motioneye, detection will be on again... :)

timdonovanuk, 2017-12-29:
I see this topic come up a lot, and y'all are making it really over complex. Motioneye shouldn't have any inbuilt presence detection - there is already a wonderful massive open source project called Home Assistant that will do stuff like this for you. 

The easiest thing to do is to use cURL to post a command to your motioneye - the same command that happens when you toggle motion on/off in the web UI. It's a bit messy because you have to send the entire config every time but it works really well.

SuperBFG7, 2017-12-29:
I agree, I was just too lazy to install, learn and configure home assistant (yet another tool) and just made the script to activate/deactivate motioneye :)

timdonovanuk, 2017-12-29:
Fair enough :) For anyone else reading I would heartily recommend Home Assistant for people looking to do presence detection, as comes with all sorts of inbuilt ways of doing this. It's also quite easy to plumb into motinoeye using curl.

sfgabe, 2018-02-12:
Part of my script at #733 might help. It rewrites the thread.conf to turn motion detection on or off (or whatever parameters you want to change). It does require a reboot to see the changes take effect, so it may not 100% solve your issue. I'm sure HomeAssistant would have worked, but like @SuperBFG7 I wanted to work with the setup I already had.

mario-g0, 2018-06-12:
@timdonovanuk  could you please point me to some instructions on how to do that? I use HA and want to activate or deactivate motion detection based on presence detection

sfgabe, 2018-06-12:
@mario-g0 - there's a bunch of ways to do presence detection through Home Assistant, it just depends on what works with your setup. 

I now have home assistant using the google location option to decide if I'm home or away (state change trigger), then sending a curl to activate the motioneyeos action button (command line activity) to turn on motion detection (using a simplified version of the script mentioned above).

Bluetooth detection to your phone or other device you carry with you is probably a more reliable trigger but I couldn't get it working in my setup.

evripidis, 2018-08-25:
@SuperBFG7 That seems to be extremely useful! Could you please provide a quick tip how to get it run?

SuperBFG7, 2018-08-26:
@evripidis sure:
1. in your WiFi router config: assign fixed IPs to the devices you want to track; I'm using my and my wifes mobilephone
1. put the script https://github.com/SuperBFG7/scripts/blob/master/pi/surveillance.sh in /usr/local/bin or anywhere you like and can execute it
1. replace the dummy IPs on line 6 with the fixed IPs of your devices
1. replace the dummy motioneye URL on line 8 with your motioneye installations host/port
1. call the script manually to test:
   1. `surveillance.sh status` to get the current status
   1.  `surveillance.sh pause` to stop motion detection
   1.  `surveillance.sh status` to start motion detection
1. call it without arguments to use the presence detection
1. create a cron job to run the script every couple of minutes

evripidis, 2018-08-26:
@SuperBFG7 Thank you very much for your quick response! The thing is that I've messed up a little bit of motioneye and motioneyeos and now I am trying to setup a motioneyeos from scratch. Unfortunately I think this script is not working on motioneyeos, or at least I am not doing it right. 
I'll keep your tips if I have to go back to my previous motioneye setup.
Thanks!

kabadisha, 2019-01-11:
@SuperBFG7 
FYI, I did something similar but found that pinging my phone was quite unreliable. If my phone was asleep, it often wouldn't respond to pings. I found that I could get my router to output the wireless association list on the command line, so I used that instead. It seems WAAAY more reliable.

Gist here: https://gist.github.com/kabadisha/a87e50311d3ed4589f17a6e37c574a44

SuperBFG7, 2019-01-14:
Yes, it is not very reliable. Thanks for sharing your solution! Unfortunately I can’t execute scripts on my router. :(
My script uses an extra loop, retrying the ping 30s later again, to mitigate this. Not very elegant, but I found it improved the reliability to a usable level.
Querying for more than one device (i.e. phone, tablet, laptop, ...) also helps. The probability that no device is reached is very low then. Of course you should not forget to turn the devices off or take them with you when you leave. :)

pantchox, 2019-01-14:
I am using pushbullet and node-red to apply activate on/of the motion
detection,
i use node-red to change the motion eye config, kill it and restart it -
takes from 10 - 20 seconds.
Also when there is a motion i get notification on my phone with pushbullet
since every motion calls  node-red REST endpoint that sends a message to
pushbullet.

I will post a blog post in the following weeks.

SuperBFG7, 2019-01-14:
For notification I am using IFTTT triggered by images uploaded to Google cloud.

SdeGeata, 2019-07-22:
@SuperBFG7:

How do i get this to work? I put the script where suggested, chmoded to make it exectuable and set the crontab. I try to run it in the command line and get this feedback:

`surveillance.sh status
\/usr/local/bin/surveillance.sh: line 3: /usr/local/bin/includes.sh: No such file or directory`

SuperBFG7, 2019-07-28:
Hi @SdeGeata

the script is part of my collection of scripts and uses a common function.
I've pushed a new version, which should remove the dependency and allow it
to be used standalone.

Just change the IPs to check (your phones, tablets, ...) and the URL where
motion(eye) is running.

Hope this helps.

Daniel

SdeGeata, 2019-07-28:
@SuperBFG7 Thanks! I'll give it a try!

SdeGeata, 2019-08-11:
@SuperBFG7 Thanks again for your help. It turns out that even once I got the thing to work (crontab execution permission issues...) it was still having a hard time detecting my iPhone properly. Gonna explore other options - perhaps buy an SSH friendly router would be a start?

piachuPL, 2019-09-13:
i think that helps: https://github.com/piachuPL/Motionoffon

Toon-nooT, 2019-09-17:
@SuperBFG7 I looked at that script. Is it possible to work only with inputting MAC addresses instead of IPs? This would remove the need to assign fixed IPs. I see you're already looking at the arp cache...

SuperBFG7, 2019-09-18:
You can skip the ping and only grep for the MAC in the arp cache, but
unless you run this on your router, I think it will work considerably worse.
The arp cache will probably only contain the MAC if there has been a
communication with the device (i.e. phone) recently.

markptrueman, 2020-02-22:
> Nope, unfortunately that won't be possible because of the working schedule management which will continuously overwrite the setting. I really must do something about this.

Have you been able to do anything about this? Would love to be able to turn off motion detection with motioneye running. Every time i tell motion to stop motion detection, motioneye continues to detect motion.

Thanks

Mark

JamBalaya56562, 2026-11-13:
motionEye 0.45.0b1 can do this while running: `POST /action/<camera id>/detectionpause` and `detectionstart` (#3468), and the working schedule keeps such a change until its next switch (#3506), which was the blocker mentioned above. From a script, log in once as in the `curl -c` / `curl -b` example and post to these URLs. There is no motionEye URL to read the state; on the motionEye machine, Motion answers `curl http://127.0.0.1:7999/<n>/detection/status`, where `<n>` is the camera's position in motion.conf.

closed 2026-11-13 by JamBalaya56562