motioneye-project/motioneye #252: Recorded and visualized video imperfections

lozziboy, 2016-07-16:
I started using motionEyeOs for raspberry PI2 recording video from two IP network cameras (RTSP) 1280x720p.
I encountered some imperfection in the video visualized by the motioneye daemon and consequently into the recorded and stored video. 
The two IP cameras are connected via ethernet cable and if i try to capture the RTSP with my laptop (VLC media player) there aren't imperfections.
I thought the reason was the poor power of the raspberry pi2 and so i upgraded my system switching it onto an ubuntu server (Amd quad core 2,1 Ghz processor + 4gb ram ddr3) but to my great surprise the issue remains. In the system monitor the CPU resource is used at 26/28 % and what i think now is that there should be something wrong in the motioneye daemon.

Here there are some pictures about the problem

[image]
[image]
[image]

lozziboy, 2016-07-20:
Problem solved!
The issue was related to the internet protocol used (UDP).
So, what i suggest is to use TCP in order to avoid any imperfection.

[image]

closed 2016-07-20 by lozziboy

ccrisan, 2016-07-20:
Hmm, you're not the first one to have a better RTSP experience with TCP. I'll make sure TCP comes as the first option from now on.