motioneye-project/motioneye #253: RTSP camera test does not support Digest authentication

jogu, 2016-07-17:
I'm having trouble setting up the amcrest ProHD 1080P IP2M-841 camera I have inside motioneye. 

The camera itself works absolutely fine with motion (I'm using MrDave's version).

The RTSP camera test (test_rtsp_url in utils.py) only handles Basic HTTP authentication, and this camera requires digest authentication.

If I make this change to motioneye (as a hack to let me proceed) then I can add the camera fine:

--- utils.py~	2016-07-05 08:30:37.000000000 +0100
+++ utils.py	2016-07-17 10:49:12.144859369 +0100
@@ -507,7 +507,8 @@
                 seek_server()

             elif data.endswith('401 '):
-                handle_error('authentication failed')
+                handle_success(None)

             else:
                 handle_error('rtsp netcam returned erroneous response: %s' % data)

It seems this was partly discussed under https://github.com/ccrisan/motioneye/issues/78 but I didn't really understand the conclusion, I think there were multiple different issues raised in that discussion.

Here's a log for what motion does when talking to the camera successfully:

T 192.168.1.8:54418 -> 192.168.1.28:554 [AP]
  OPTIONS rtsp://192.168.1.28:554/cam/realmonitor?channel=1&subtype=1 RTSP/1.
  0..CSeq: 1..User-Agent: Lavf56.25.101....
##
T 192.168.1.28:554 -> 192.168.1.8:54418 [AP]
  RTSP/1.0 401 Unauthorized..CSeq: 1..WWW-Authenticate: Digest realm="Login t
  o AMC0007RVY5EQ6DV29", nonce="5ae62bb61f1fXXX08e0fb833c0747f60", ....
##
T 192.168.1.8:54418 -> 192.168.1.28:554 [AP]
  OPTIONS rtsp://192.168.1.28:554/cam/realmonitor?channel=1&subtype=1 RTSP/1.
  0..CSeq: 2..User-Agent: Lavf56.25.101..Authorization: Digest username="admi
  n", realm="Login to AMC0007RVY5EQ6DV29", nonce="5ae62bb61f1f44d08e0fb833c07
  47f60", uri="rtsp://192.168.1.28:554/cam/realmonitor?channel=1&subtype=1",
  response="e5fa09a3XXX0e6fa6c39f70faac8f367"....

ie. first it tries without the username & password, then it looks at the response, find it needs digest auth and then tries against with digest auth.

ccrisan, 2016-07-18:
What happened in https://github.com/ccrisan/motioneye/issues/78 is that people repeatedly requested auth support for RTSP and I chose to implement the Basic scheme.

Indeed, it's only a test performed by motionEye before passing the parameters to motion, so if motionEye would just let any RTSP urls pass, ignoring 401 responses, chances are that they would actually work with motion.

The reason why I did not implement the Digest auth scheme when testing the URL is that it was too complicated :). The reason why I hesitate to merge your workaround into master is that it would pass even if you supplied invalid credentials, which is one of the points of the  testing process.

jogu, 2016-07-19:
That all makes sense, thanks. I wasn't suggesting merging my workaround, patch was more just to share exactly how I got it working.

I *think* the failure response might include the information that digest auth is required. If so I could try doing a pull request that just allows the camera to be configured anyway if the camera tells us it requires digest auth, but still checks the username/password are correct for cameras that do basic auth. Not ideal. I'm guessing actually implementing digest auth would take me significantly longer.

jogu, 2016-08-13:
I checked; the failure message motioneye gets back doesn't include any hint that only Digest authentication is acceptable.

I think motioneye needs to follow a similar path to motion - to always try the request without authentication first. If the response says to use basic auth, try that, but if it says digest auth (until/unless it's implemented) treat that as success. I don't think I have time to try implementing that.

ccrisan, 2016-08-14:
@jogu I agree. Given that motion accepts RTSP digest authentication, the UI should not stand in the way of adding such a camera.

The biggest challenge in implementing RTSP authentication schemes on motionEye's side is that I cannot seem to simulate authentication-enabled RTSP cameras. At least not with VLC. I usually ask people to temporarily give me access to their RTSP cams (via IP forwarding or so) so I can see what's going on and test my code.

This being said, I'd be happy to change the code in such a way that Digest RTSP cameras are always allowed to "pass", but I'd need such a camera to implement this. Would you by any chance have one at your disposal and would be willing to temporarily provide me with access?

jogu, 2016-08-14:
I think ultimately perhaps the way this should work is that you should be able to use motion (or some utility based on motion's code) to test camera settings and perhaps even retrieve a still frame from the camera. I don't think this would be easy to achieve, given the way the motion code is currently structured.

I'll email you an rtsp url you can use to test (to the email in your github profile) in the next few minutes.

ccrisan, 2016-08-14:
https://github.com/ccrisan/motioneye/commit/cdb070d767a8a979d448a7443a14536f14acb2d3 should fix this. It will simply allow you to add any Digest rtsp camera, regarding of credentials. Basic authentication is still handled as it should.

Please test this and let me know if it works as expected. And thanks for providing the RTSP link!

jogu, 2016-08-19:
Code change looks good to me, thanks! It'll probably be a while before I get a chance to try it, so feel free to close this - I'm too busy looking at other things :-)

closed 2016-08-20 by ccrisan