motioneye-project/motioneye #254: Can the new resolutions from raspberry camera be added to motioneye gui?

ArmandH, 2016-07-17:
Hi,

First of all great work you do here, love it, just got it running on raspberry pi zero.
I have the latest v2.1 camera which supports different resolutions, see:
https://www.raspberrypi.org/blog/new-camera-mode-released/

If it possible to get these added to the motioneye gui as selection, maybe only visible when you have this type of camera.
or a custom option that add the resolution to config file?
I am specially interested in the new resolution that uses the GPU instead of the CPU as the CPU load is pretty high on the zero

gtr33m, 2016-07-27:
Not specifically for the raspberry pi camera, but I too would like some new resolutions added to the GUI.  Changing them in the thread-#.conf files works fine, but if you use the GUI to change anything else, they get overwritten.

Specifically one of my cameras uses 360x640

ArmandH, 2016-07-28:
Ok thanks i have a look at the thread.conf file under /etc/motioneyeos/
for others some options explained here: http://www.lavrsen.dk/foswiki/bin/view/Motion/ConfigFileOptions

and yes i agree there can be some more options in the GUI for resolutions as well and i would like to see fast network camera option being available when you install motioneye on top of OS instead of using the image.

gtr33m, 2016-07-28:
For those who have it installed on an OS like Raspbian, the thread files are in/etc/motioneye

efelon, 2016-09-09:
To tackle all the request, a possibility would be nice to manually add custom resolutions to motioneye.conf, which will be available in the GUI after restart.

ozett, 2016-10-06:
👍 
(missing full-hd resolution ..)

ccrisan, 2016-11-16:
Fixed with https://github.com/ccrisan/motioneye/commit/f1298c012312aee69e231d2831abab2c3d76a0a6 and https://github.com/ccrisan/motioneye/commit/d61858b0ad47a13428e850ee475619ac57e0af62.

closed 2016-11-16 by ccrisan