motioneye-project/motioneye #255: cant add camera

damparker, 2016-07-18:
Hi, I am trying to add 2 I.P cameras to my raspberry pi running motioneye. Both cameras can be seen on my home network, however when I try and add either it comes up with an error "unsupported camera" I have tried using network camera as well as MJPEG camera.  Here is the string I am entering http://192.168.1.14:80. I have tried different ports with no success. I have a USB

ccrisan, 2016-07-18:
The mjpeg URL is usually more complex than your URL, which is probably the URL for your camera's web UI. You need to find the correct mjpeg URL for your camera.

damparker, 2016-07-18:
Ok thanks. Sorry nub question. Do i just google to find the url?

ektus, 2016-08-21:
That's the usual way to do it. Search for the streaming url of your camera make and type. Or use something like onvifdm (ONVIF device manager) to view it's properties like has been outlined in #109

Marijn0, 2026-10-27:
Closing this old issue as per the discussion above.

closed 2026-10-27 by Marijn0