motioneye-project/motioneye #256: sendmail is clogging up, email notification stops

fireheadman, 2016-07-19:
Not sure what is causing this....  I have noticed the past 2 days notifications have stopped.  when I look at the system, I see sendmail clogged up

pi@webcam01:~ $ ps aux | grep sendmail | wc -l
58

Some of the output:
pi@webcam01:~ $ ps aux|grep motion
root       368  0.0  0.0   1912   396 ?        Ss   10:28   0:00 sh -c /usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" start 1; /usr/bin/python /usr/local/lib/python2.7/dist-packages/motioneye/meyectl.pyc sendmail -c /etc/motioneye/motioneye.conf 'smtp.<myisp>.net' '587' 'username@<myisp>.net' '<myPASSWORD>' 'True' 'motion@<myDOMAIN>.com' '<myEMAIL>' 'motion_start' '1' '2016-07-19T10:28:10' '5'  &
root       371  0.1  1.2  19892 11776 ?        S    11:27   0:02 /usr/bin/python /usr/local/lib/python2.7/dist-packages/motioneye/meyectl.pyc sendmail -c /etc/motioneye/motioneye.conf 'smtp.<myisp>.net' '587' 'username@<myisp>.net' '<myPASSWORD>' 'True' 'motion@<myDOMAIN>.com' '<myEMAIL> motion_start 2 2016-07-19T11:27:01 5
root       404  1.4  2.2  58116 21720 ?        Ssl  Jul18  10:57 /usr/bin/python /usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf
root       466  0.0  1.4  18376 13920 ?        S    10:28   0:01 /usr/bin/python /usr/local/lib/python2.7/dist-packages/motioneye/meyectl.pyc sendmail -c /etc/motioneye/motioneye.conf smtp.<myisp>.net' '587' 'username@<myisp>.net' '<myPASSWORD>' 'True' 'motion@<myDOMAIN>.com' '<myEMAIL> motion_start 1 2016-07-19T10:28:10 5
root       752  0.0  1.3  27512 12732 ?        S    Jul18   0:00 /usr/bin/python /usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf

fireheadman, 2016-07-19:
Was looking at the sendmail.py script.   I was under the impression the actual "sendmail" program was being called out, but it looks like you have python doing the email work.

```
2016-07-19 12:01:58: [sendmail]     INFO: email sent
2016-07-19 12:04:55: [sendmail]     INFO: email sent
2016-07-19 12:13:37: [sendmail]     INFO: email sent
2016-07-19 12:25:30: [sendmail]     INFO: email sent
2016-07-19 12:25:43: [sendmail]    ERROR: timeout waiting for the media listing process to finish
2016-07-19 12:25:45: [sendmail]     INFO: email sent
2016-07-19 12:52:05: [sendmail]    ERROR: timeout waiting for the media listing process to finish
2016-07-19 12:52:07: [sendmail]     INFO: email sent
2016-07-19 12:52:18: [sendmail]     INFO: email sent
```

From the time it took me to write the first message in this thread to this one... I have accumulated 3 more stale/clogged emails in queue.  When I look at the log (/var/log/motion.log), I see 1 of them matching (timestamp = 12:25.

```
pi@webcam01:~/domoticz $ ps aux|grep sendmail
pi        5255  0.0  0.2   4276  2000 pts/0    S+   13:07   0:00 grep --color=auto sendmail
root     16173  0.0  0.0   1912   388 ?        Ss   12:25   0:00 sh -c /usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" start 2; /usr/bin/python /usr/local/lib/python2.7/dist-packages/motioneye/meyectl.pyc sendmail -c /etc/motioneye/motioneye.conf 'smtp.<myisp>.net' '587' 'username@<myisp>.net' '<myPASSWORD>' 'True' 'motion@<myDOMAIN>.com' '<myEMAIL>' 'motion_start' '2' '2016-07-19T12:25:25' '5'  &
root     16267  0.0  1.4  18376 13988 ?        S    12:25   0:01 /usr/bin/python /usr/local/lib/python2.7/dist-packages/motioneye/meyectl.pyc sendmail -c /etc/motioneye/motioneye.conf smtp.<myisp>.net' '587' 'username@<myisp>.net' '<myPASSWORD>' 'True' 'motion@<myDOMAIN>.com' '<myEMAIL> motion_start 2 2016-07-19T12:25:25 5
root     17261  0.1  1.2  20148 11732 ?        S    12:25   0:03 /usr/bin/python /usr/local/lib/python2.7/dist-packages/motioneye/meyectl.pyc sendmail -c /etc/motioneye/motioneye.conf smtp.<myisp>.net' '587' 'username@<myisp>.net' '<myPASSWORD>' 'True' 'motion@<myDOMAIN>.com' '<myEMAIL> motion_start 2 2016-07-19T12:25:25 5
```

ccrisan, 2016-07-19:
1. Please attach your motion.log file here. It usually contains more details about email sending.
2. Ideally the email sending process should disappear in the near future being completely embedded in the motionEye process.

fireheadman, 2016-07-19:
/var/log/motion.log
```
pi@webcam01:~/domoticz $ cat /var/log/motion.log
[0] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
[0] [NTC] [ALL] config_thread: Processing config file thread-1.conf
[0] [NTC] [ALL] config_thread: Processing config file thread-2.conf
[0] [NTC] [ALL] motion_startup: Motion Unofficial-Git-UNKNOWN Started
[0] [NTC] [ALL] motion_startup: Logging to syslog
[0] [NTC] [ALL] motion_startup: Using default log type (ALL)
[0] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
[1] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD
[2] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD
2016-07-19 12:01:58: [sendmail]     INFO: email sent
2016-07-19 12:04:55: [sendmail]     INFO: email sent
2016-07-19 12:13:37: [sendmail]     INFO: email sent
2016-07-19 12:25:30: [sendmail]     INFO: email sent
2016-07-19 12:25:43: [sendmail]    ERROR: timeout waiting for the media listing process to finish
2016-07-19 12:25:45: [sendmail]     INFO: email sent
2016-07-19 12:52:05: [sendmail]    ERROR: timeout waiting for the media listing process to finish
2016-07-19 12:52:07: [sendmail]     INFO: email sent
2016-07-19 12:52:18: [sendmail]     INFO: email sent
```

ccrisan, 2016-07-19:
It seems that you have too many media files gathered and it takes too much to parse them all (hence the errors you see). Either clean them up and configure the camera in such a way that no more than, say, a few hundred pictures are normally stored in your camera's root folder, or set the email picture timespan to 0 to disable picture email attachments.

fireheadman, 2016-07-19:
I would agree on the "too much" piece.
I use the Still Image part of motioneye so that I can get a snapshot attached to the notification email since there is not a way to attached the actual video itself.

I used to attach my videos to email when I used the original "motion" program.

However, once a motion event occurs, this causes 50+ Still Images to be saved, so just having 5 events trigger within a few hours can cause 250+ images to piles up.    I have already created a cron job to clean up anything over 900min (15hrs of age).

fireheadman, 2016-07-19:
Here is my cleanup solution:
```
# --------------min (0 - 59)
# | ------------hour (0 - 23)
# | | ----------day of month (1 - 31)
# | | | --------month (1 - 12)
# | | | | ------day of week (0 - 6) (Sunday=0)
# | | | | |
# * * * * *     command to be executed
##
# FIND/DELETE CAMERA IMAGE FILES OLDER THAN 1HR EVERY 10 MIN
*/10 * * * * find /data/webcam01/video_files/cam*/*/images -type f -mmin +60 -exec rm -f {} \;
```

JamBalaya56562, 2026-11-07:
Closing as answered: as ccrisan found above, too many media files made each notification take too long, and keeping fewer files, as with your cleanup script, avoids it.

closed 2026-11-07 by JamBalaya56562