motioneye-project/motioneye #257: Action Buttons

Hard2findaNameHere, 2016-07-21:
When can more action buttons be included into MotionEye?  The current six are good but another six would be fantastic ie, Up_1, Down_1, Left_1, Right_1, Forward_1 and Back_1

yanoschik, 2016-07-21:
done & discussed [here](https://github.com/ccrisan/motioneye/commit/834a68233f8c162b0c77049cbd177ee053876f27)

Hard2findaNameHere, 2016-07-21:
Could you explain a little more please? I'm not understanding all those lines of code...

----- Reply message -----

https://github.com/ccrisan/motioneye/issues/257#issuecomment-234258820

hexordecimal, 2019-11-26:
i placed my scripts in /etc/motioneye/ then made them executable `sudo chmod +x down_1` for each script, then ran `sudo service motioneye restart,` or `sudo reboot`, that should resolve it.