motioneye-project/motioneye #258: Changing IP Address

Hard2findaNameHere, 2016-07-21:
I have a dynamic IP address to my MotionEye Raspberry Pi which changes regularly making it difficult or impossible to remotely log back into it.  Is there anyway MotionEye can include the IP Address (http) when issuing an alert?

I have tried other third party websites like NOIP but the MotionEye image does not have the utilities like apt-get,  make to install their client software that notifies of any IP change.

GeertvanWissen, 2016-08-04:
You can see the IP address from where the notification mail had been send in the e-mail headers. But you're right, that isn't the most ellagant solution. Better would be to include a hyperlink in the mail body.

JamBalaya56562, 2026-11-13:
As @ccrisan said on the same request in #212, this is outside motionEye: a dynamic DNS client on the device or the router keeps a fixed name pointing at the changing address, and the notification emails then only need that name. Closing.

closed 2026-11-13 by JamBalaya56562