motioneye-project/motioneye #259: Message: An error occurred. Refreshing is recommended.

lozziboy, 2016-07-22:
Hello Calin,
after three weeks of perfect working i got this error in the motioneye service.
I i try to refresh the page the problem appears again and no video are visualized.
What can i do in order to help you find the reason?

[image]

ArmandH, 2016-07-28:
Maybe something was hanging, motioneyeos has watchdog enabled by default, so could be that your OS has rebooted.

Checkout if you find anything related in the troubleshooting section:
https://github.com/ccrisan/motioneyeos/wiki/Troubleshooting

luetzel, 2016-08-03:
I had exactly the same problem. No way to add one of my remote MotionEye Cams using the 'Add Camera' dialog, although I was able to reach the camera's control interface. However, I was able to fix this issue, by copying and editing /etc/motioneye/thread-2.conf, changing both # @id and # @host.
Only one out of four camera devices was affected, all having 0.33 installed.

woodboy12, 2021-03-11:
I had this problem. Check your storage space available. Mine was completely full. Motioneye started giving me errors when trying to adding cameras and ssh'ing into the system. I started from scratch with a new SD card and everything seems to work fine now.

starbasessd, 2021-03-11:
@woodboy12 This issue was opened 5 years ago. I doubt it is being followed now.

closed 2021-03-11 by starbasessd