motioneye-project/motioneye #260: CSI camera not recognised, unable to load Module bcm2835-v412.

MauriceTurner, 2016-07-23:
Hello Calin,

I’m having difficulties getting my Raspberry Pi CSI Camera to be detected by motioneye.
I have today installed the latest version of Raspbian from the foundation website:
RASPBIAN JESSIE
Full desktop image based on Debian Jessie
Version:
May 2016
Release date:
2016-05-27
Kernel version:4.4

Whist installing motioneye, I took note of your instruction:
If you want to use the CSI camera module for the Raspberry PI, you need to add bcm2835-v4l2 to /etc/modules and reboot.

Having competed the installation instructions I was able to connect to motioneye via http, however motioneye was unable to detect the camera module.

I can confirm that I can capture still images using the raspistill command.

My limited troubleshooting skills led me to try the following command:
sudo modprobe bcm2835-v412, which returned FATAL: Module bcm2835-v412.

Having failed to resolve the issue I have now reformatted the SD card, reinstalled Jessie and thought now would be a good time to pause and seek assistance.

Do you have any suggestions on what I can try next please?

MauriceTurner, 2016-07-23:
sudo modprobe bcm2835-v412, returned FATAL: Module bcm2835-v412 not found

ccrisan, 2016-07-23:
It's V4*L*2, not V412!

MauriceTurner, 2016-07-24:
Thank you - so it's time to get my prescription checked then!

closed 2016-07-24 by MauriceTurner

ccrisan, 2016-07-24:
:)

MauriceTurner, 2016-07-24:
Thanks for quick response. I hope to try again I’ll let you know how I get on.

Regards,

Maurice.

astechnolabs-html, 2021-11-09:
Still not working

starbasessd, 2021-11-09:
4.5 years later? Really?
I'm closing this issue. Please open a new issue, include your setup:
OS used
Instructions for motionEye installation.
Which board used
Which camera used
Does the camera work with raspistill or raspivid?
If using RaspberryPiOS, is it Buster or Bullseye?
If the CSI Camera, is it enabled in raspi-config?
Are you installing the camera as MMAL (NOT V4L2)?