motioneye-project/motioneye #262: Frame Rate Low

LuffyyyffuL, 2016-07-24:
Hello,

When I set "Frame Rate" parameter to 20 for example, i have a slow picture : 3fps instead of 20fps.

My Raspberry Pi is connected to Ethernet port 100 Mbit/s.

Why ? How to do to optimized ?

Thank you

Anthony

ArmandH, 2016-07-28:
Try lowering your resolution then you will see you get higher frame rates or switch to the fast network camera, works a lot better then motion, but.... for this you do have to use the motioneyeos, if you have installed it on top of an os then the option is not there i do hope that Calin will put that option in the normal install as well as it would be very welcome.

More info here: https://github.com/ccrisan/motioneyeos/wiki/Fast-Network-Camera

jasaw, 2017-07-28:
@LuffyyyffuL What's your configuration? Are you saving the video into a movie on your Pi?

JamBalaya56562, 2026-11-07:
Closing, as jasaw's question about the configuration went unanswered. As ArmandH said above, a lower resolution gives a higher frame rate, and on motionEyeOS the Fast Network Camera mode streams with less load. Please open a new issue with the Pi model and `motion.log` if the frame rate is still far below the setting with a current release.

closed 2026-11-07 by JamBalaya56562