motioneye-project/motioneye #263: relayevent.sh: Permission denied

ChriZathens, 2016-07-25:
Hello!
Just noticed that in motion.log I get this:

> File of type 2 saved to: /var/www/downloads/motioneye/Year-2016/Month-07/Day-25/Hour-10/104600-snapshot.jpg
>  &: /usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh: Permission denied

This is a snapshot set to be taken every minute.
The image gets stored normally - but what is this permission denied I don't know..

ccrisan, 2016-07-25:
Well I have no idea how your relayevent.sh script became non-executable. How did you install/upgrade your motionEye setup? What version do you run?

ChriZathens, 2016-07-26:
Hello, Calin and thanks for the reply.
I initially installed using 

> pip install motioneye

And later upgraded using 

>  pip install motioneye --upgrade

as per the wiki
As for my version:

Current Version	0.32.2
Regarding releayevent.sh, these are the perms, so no executable.

> root@server:/usr/local/lib/python2.7/dist-packages/motioneye/scripts# ls -l
total 4
-rw-r--r-- 1 root staff 1093 Jul 10 18:48 relayevent.sh

So do I simply need to add +x to it?

ccrisan, 2016-07-26:
Yeap, that should do it. The file in the pip archive has the executable flag set so there's something else that must have happenede there.

ChriZathens, 2016-07-26:
OK added a +x flag and it seems to work (no more log messages)
I am guessing the thumbs problem is a different issue and I should open a different ticket?

ccrisan, 2016-07-26:
Most likely.

ChriZathens, 2016-08-16:
Hello again, calin..
Reopening this, because just noticed that after the update to 0.33 the problem came back.
releayevent.sh is no more executable.
I added a +x flag again and seems to start working again.
And just to confirm, I just upgraded to 0.33.1 and the new releayevent.sh does no have the +x flag

ccrisan, 2016-08-16:
How do you install/upgrade your motionEye? If you go the pip/pypi way, you'll probably end up downloading the file from [here](https://pypi.python.org/pypi?:action=display&name=motioneye&version=0.33.1).

I ran the following commands to check the permissions on the `relayevent.sh` script:

    wget https://pypi.python.org/packages/1b/86/eb7f0f6befa022f4cf03596b17fdd89d57e172ece13414969b0db9d7394b/motioneye-0.33.1.tar.gz
    tar tvf motioneye-0.33.1.tar.gz  | grep 'relayevent.sh'

And it says:

    -rwxr-xr-x ccrisan/ccrisan  1093 2016-07-31 14:47 motioneye-0.33.1/motioneye/scripts/relayevent.sh

Is there something your pip (or pip in general) does that messes up permissions? Could you investigate this further? Maybe the pip upgrade log could help.

ChriZathens, 2016-08-16:
Hello again..
This is the pip.log file`/usr/bin/pip run on Tue Aug 16 16:30:12 2016
Downloading/unpacking motioneye
  Getting page http://pypi.python.org/simple/motioneye
Exception:
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/pip/basecommand.py", line 126, in main
    self.run(options, args)
  File "/usr/lib/python2.7/dist-packages/pip/commands/install.py", line 223, in run
    requirement_set.prepare_files(finder, force_root_egg_info=self.bundle, bundle=self.bundle)
  File "/usr/lib/python2.7/dist-packages/pip/req.py", line 948, in prepare_files
    url = finder.find_requirement(req_to_install, upgrade=self.upgrade)
  File "/usr/lib/python2.7/dist-packages/pip/index.py", line 97, in find_requirement
    page = self._get_page(main_index_url, req)
  File "/usr/lib/python2.7/dist-packages/pip/index.py", line 324, in _get_page
    return HTMLPage.get_page(link, req, cache=self.cache)
  File "/usr/lib/python2.7/dist-packages/pip/index.py", line 441, in get_page
    resp = urlopen(url)
  File "/usr/lib/python2.7/dist-packages/pip/download.py", line 83, in __call__
    response = urllib2.urlopen(self.get_request(url))
  File "/usr/lib/python2.7/urllib2.py", line 126, in urlopen
    return _opener.open(url, data, timeout)
  File "/usr/lib/python2.7/urllib2.py", line 400, in open
    response = self._open(req, data)
  File "/usr/lib/python2.7/urllib2.py", line 418, in _open
    '_open', req)
  File "/usr/lib/python2.7/urllib2.py", line 378, in _call_chain
    result = func(*args)
  File "/usr/lib/python2.7/urllib2.py", line 1207, in http_open
    return self.do_open(httplib.HTTPConnection, req)
  File "/usr/lib/python2.7/urllib2.py", line 1174, in do_open
    h.request(req.get_method(), req.get_selector(), req.data, headers)
KeyboardInterrupt
`
I also run the following command 

> pip install motioneye --upgrade --verbose --log install.log
which created a much larger install.log file. In case you need it I can attach it here.

ChriZathens, 2016-08-16:
Later update: Up until now I was using python-pip version 1.0 from ubuntu repository
Everytime I was running the command pip install motioneye --upgrade it seemed to download and install motioneye from the beginning. It seemed strange, but thought it was the intented behavior 
So I did the following:

> curl "https://bootstrap.pypa.io/get-pip.py" -o "get-pip.py"

> ./get-pip.py

But afterwards after running " pip install motioneye --upgrade" I got pip not found in /usr/bin/pip
So I reinstalled the version from the ubuntu repo and then trying to upgrade motioneye I got this:

> pip install motioneye --upgrade
Requirement already up-to-date: motioneye in /usr/local/lib/python2.7/dist-packages
Requirement already up-to-date: tornado>=3.1 in /usr/local/lib/python2.7/dist-packages (from motioneye)
Requirement already up-to-date: jinja2 in /usr/local/lib/python2.7/dist-packages (from motioneye)
Requirement already up-to-date: pillow in /usr/local/lib/python2.7/dist-packages (from motioneye)
Requirement already up-to-date: pycurl in /usr/local/lib/python2.7/dist-packages (from motioneye)
Requirement already up-to-date: backports.ssl_match_hostname in /usr/local/lib/python2.7/dist-packages (from tornado>=3.1->motioneye)
Requirement already up-to-date: singledispatch in /usr/local/lib/python2.7/dist-packages (from tornado>=3.1->motioneye)
Requirement already up-to-date: certifi in /usr/local/lib/python2.7/dist-packages (from tornado>=3.1->motioneye)
Requirement already up-to-date: backports_abc>=0.4 in /usr/local/lib/python2.7/dist-packages (from tornado>=3.1->motioneye)
Requirement already up-to-date: MarkupSafe in /usr/local/lib/python2.7/dist-packages (from jinja2->motioneye)
Requirement already up-to-date: six in /usr/local/lib/python2.7/dist-packages (from singledispatch->tornado>=3.1->motioneye)
You are using pip version 1.0, however version 8.1.2 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.

which seems more correct to me (since it no more tried to install motioneye from the beginning, it rather realized that it is up to date)
So I run now "pip install --upgrade pip" and I got again to the point where it cannot find pip in /usr/bin/pip.
So finally created a symlink from /usr/local/bin/pip to /usr/bin/pip and seems to be working now..
Running pip --version yields: 

> pip 8.1.2 from /usr/local/lib/python2.7/dist-packages (python 2.7)
while before I was getting
> pip 1.0 from /usr/lib/python2.7/dist-packages (python 2.7)

So I guess we will see how it goes from here

closed 2016-08-16 by ChriZathens