motioneye-project/motioneye #264: Wireless

gmonk63, 2016-07-26:
Wireless config very unstable initial config works fine untill  I access the device and make a change which requires a reboot. After that wireless stops working

ccrisan, 2016-07-26:
Can you please attach your log files? Also some more info regarding the hardware you're using would be useful.

gmonk63, 2016-07-28:
Sorry for the late response.. Its seem to be working now. I was 
attempting to edit the wpa_supplicant.conf file right after imaging so i 
could just boot up and connect via wifi but it does not seem to like 
that since every time I would access the interface for the initial 
config  then make a change i would loose connectivity after reboot.  I 
ended up just setting it up through the eth0 first  then set up the wifi 
through the gui instead and it seems to be stable doing it that way.  I 
saw the same issues on both my RPI2 and RP1 B+

Thanks

Gary

ArmandH, 2016-07-28:
Wifi is not so stable on the Pi maybe you have issues with that, you can google there are several solutions to restart the wifi when it hangs and looses network or other solutions to reboot the pi when it hangs through watchdog or by using a remote login every so many hours.

ccrisan, 2017-08-05:
If this is about motionEyeOS, many things have been fixed lately with regards to the WiFi connection.

closed 2017-08-05 by ccrisan