motioneye-project/motioneye #265: No thumbs file created automatically

ChriZathens, 2016-07-26:
Until recently I remember that for every motion video that was created, motioneye was automatically generating a thumbs file.
But in the past few days I realized that the thumbs images are being created only if I visit the motioneye webui and click to open movies browser.
Then, when I select a date, thumbs are being generated and I normally see thumbs on the movies browser list.
Is there something I should check? a setting maybe? (I recently changed the camera and had to redo the configuration of the camera from the beginning - did I perhaps miss something ?)
Thanks a lot again!

ChriZathens, 2016-07-26:
Additional info:
I enabled debug and accessing movies browser I see the below messages which seem absolutely normal to me

> running command "ffmpeg -i /path/to/long/directory/motioneye/Year-2016/Month-07/Day-25/Cam_Movement/Hour-22/25-07-2016_220905.avi -f mjpeg -vframes 1 -ss 4 -y /path/to/long/directory/motioneye/Year-2016/Month-07/Day-25/Cam_Movement/Hour-22/25-07-2016_220905.avi.thumb"

This happens when accessing the movies browser. I see the thumb is getting generated. But they were previously generated without needing to access the movies browser..

ccrisan, 2016-07-26:
Does the issue persist after making the `relayevent.sh` script executable? I might have been wrong on the other thread: the script actually has some impact on the generation of the thumbs. Can you please check if that is the case?

ChriZathens, 2016-07-26:
Well, it seems to be resolved - I checked some videos from 30 minutes ago and they do have thumbs files. 
So it seems that was the issue.
But for the past avi videos there are no thumb files.. Is there a way I  can run something to generate all missing thumbs????

ccrisan, 2016-07-26:
You'll need to figure out the ffmpeg command for that and run it manually or using a bash script.

closed 2016-07-26 by ccrisan

ChriZathens, 2016-07-26:
I figured it out - thanks!