motioneye-project/motioneye #266: monitor script

malvar0929, 2016-07-29:
I love this program. I'm running a couple of Pi3s and for the life of me I can't figure out how to display temperature and CPU load with the monitor command. I copied the sample script (named it monitor_1 and placed it in config folder) in the wiki and made it executable but nothing is displayed. I have the action buttons working great though. I'm very new to programming and so I'm still trying to wrap my head around basic stuff. Thank you in advance.

ccrisan, 2016-07-29:
What version is this? Have you tried to run the command manually (as the user with which motionEye runs)? Does it work as expected?

malvar0929, 2016-07-29:
I have ran the command manually and it outputs the temperature but it won't show on the camera image. I'm running the latest version 20160605 for the Pi3. Would you happen to have a script to display CPU load? Maybe I could try that. Again awesome work that you're doing. Thank you again.

ccrisan, 2016-07-29:
You'll need to use the latest prerelase for that to work. The feature wasn't present when 20160605 was released.

malvar0929, 2016-07-29:
I just updated (oops I thought I was on the latest release) and it's working great now.  One more thing  I would love to see the CPU load also. Could you point me in the right direction for a script. Thank you so much.

ccrisan, 2016-07-30:
Following is the script used by motionEyeOS on the Pi to display various stuff, including system load:

    #!/bin/bash
    net_tmp=/tmp/netspeed.tmp 
    temp=$(($(cat /sys/devices/virtual/thermal/thermal_zone0/temp) / 1000))
    load=$(cat /proc/loadavg | cut -d ' ' -f 2)
    recv=$(cat /proc/net/dev | grep -v 'lo:' | tr -s ' ' | cut -d ' ' -f 3 | tail +3 | awk '{s+=$1} END {print s}')
    send=$(cat /proc/net/dev | grep -v 'lo:' | tr -s ' ' | cut -d ' ' -f 11 | tail +3 | awk '{s+=$1} END {print s}')
    total=$(($recv + $send))
    if [ -e $net_tmp ]; then
        prev_total=$(cat $net_tmp)
        speed=$(($total - $prev_total))
    else
        speed=0
    fi
    echo $total > $net_tmp
    speed=$(($speed / 1024))
    echo -n "$temp&deg;|$load|${speed}kB/s"

closed 2016-07-30 by ccrisan

yanoschik, 2016-08-01:
there is a small error:
```
recv=$(cat /proc/net/dev | grep -v 'lo:' | tr -s ' ' | cut -d ' ' -f 3 | tail **-n** +3 | awk '{s+=$1} END {print s}')
send=$(cat /proc/net/dev | grep -v 'lo:' | tr -s ' ' | cut -d ' ' -f 11 | tail **-n** +3 | awk '{s+=$1} END {print s}')

```

ccrisan, 2016-08-01:
Hmm you might be right. The busybox tail version works w/o the `-n` argument. I'll add it anyway.

ccrisan, 2016-08-24:
fixed with 30e6657d0e97024f8848449eb1c3aba9bfc130bd