motioneye-project/motioneye #267: Motioneye fails to start on authorization failed error

kc1212, 2016-07-30:
Hi, it seems motioneye fails to start when there's an authorization error. I believe it's better if the frontend can catch the error and report it.

I guess the workaround is to edit/delete `/etc/motioneye/thread-3.conf`. Is this safe to do?

```
Jul 30 17:26:26 mynvr meyectl[3210]:     INFO: hello! this is motionEye server 0.32.2
Jul 30 17:26:26 mynvr motion[3220]: [34197888] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
Jul 30 17:26:26 mynvr motion[3220]: [34197888] [NTC] [ALL] config_thread: Processing config file thread-1.conf
Jul 30 17:26:26 mynvr motion[3220]: [34197888] [NTC] [ALL] config_thread: Processing config file thread-2.conf
Jul 30 17:26:26 mynvr motion[3220]: [34197888] [NTC] [ALL] config_thread: Processing config file thread-3.conf
Jul 30 17:26:26 mynvr motion[3220]: [34197888] [ALR] [ALL] conf_cmdparse: Unknown config option "rtsp_uses_tcp"
Jul 30 17:26:26 mynvr motion[3220]: [34197888] [NTC] [ALL] motion_startup: Motion 3.2.12+git20140228 Started with SDL support
Jul 30 17:26:26 mynvr motion[3220]: [34197888] [NTC] [ALL] motion_startup: Logging to syslog
Jul 30 17:26:26 mynvr motion[3220]: [34197888] [NTC] [ALL] motion_startup: Using default log type (ALL)
Jul 30 17:26:26 mynvr motion[3220]: [34197888] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
Jul 30 17:26:26 mynvr motion[3220]: [1] [ALR] [NET] netcam_start: Network Camera thread starting... for url (http://192.168.1.101/video.cgi)
Jul 30 17:26:26 mynvr motion[3220]: [3] [ALR] [NET] netcam_start: Network Camera thread starting... for url (rtsp://192.168.1.103/live1.sdp)
Jul 30 17:26:26 mynvr motion[3220]: [2] [ALR] [NET] netcam_start: Network Camera thread starting... for url (http://192.168.1.102/video.cgi)
Jul 30 17:26:26 mynvr motion[3220]: [2] [ALR] [NET] netcam_handler_loop: Camera handler thread [4] started
Jul 30 17:26:26 mynvr motion[3220]: [1] [ALR] [NET] netcam_handler_loop: Camera handler thread [5] started
Jul 30 17:26:26 mynvr motion[3220]: [1] [WRN] [NET] netcam_next: called with no data in buffer
Jul 30 17:26:27 mynvr motion[3220]: [3] [ALR] [NET] rtsp_connect: unable to open input(rtsp://192.168.1.103:554/live1.sdp): -825242872 - Server returned 401 Unauthorized (authorization failed)
Jul 30 17:26:28 mynvr meyectl[3210]: Traceback (most recent call last):
Jul 30 17:26:28 mynvr meyectl[3210]:   File "/usr/local/bin/meyectl", line 11, in <module>
Jul 30 17:26:28 mynvr meyectl[3210]:     sys.exit(main())
Jul 30 17:26:28 mynvr meyectl[3210]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/meyectl.py", line 257, in main
Jul 30 17:26:28 mynvr meyectl[3210]:     server.main(arg_parser, sys.argv[2:], command[:-6])
Jul 30 17:26:28 mynvr meyectl[3210]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/server.py", line 434, in main
Jul 30 17:26:28 mynvr meyectl[3210]:     run()
Jul 30 17:26:28 mynvr meyectl[3210]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/server.py", line 367, in run
Jul 30 17:26:28 mynvr meyectl[3210]:     start_motion()
Jul 30 17:26:28 mynvr meyectl[3210]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/server.py", line 334, in start_motion
Jul 30 17:26:28 mynvr meyectl[3210]:     motionctl.start()
Jul 30 17:26:28 mynvr meyectl[3210]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/motionctl.py", line 128, in start
Jul 30 17:26:28 mynvr meyectl[3210]:     raise Exception('motion failed to start')
Jul 30 17:26:28 mynvr meyectl[3210]: Exception: motion failed to start
Jul 30 17:26:28 mynvr systemd[1]: motioneye.service: Main process exited, code=exited, status=1/FAILURE
Jul 30 17:26:28 mynvr systemd[1]: motioneye.service: Unit entered failed state.
Jul 30 17:26:28 mynvr systemd[1]: motioneye.service: Failed with result 'exit-code'.
```

vedranius, 2019-03-02:
I'm having an exact error. 

Any solution?

Thanks!

JamBalaya56562, 2026-11-08:
Closing, as this was with motionEye 0.32.2 and Motion 3.2.12: the log shows Motion exiting when an RTSP camera answered `401 Unauthorized`, and motionEye stopping because Motion did not start. Current Motion releases keep retrying a camera that fails to connect instead of exiting, so the other cameras and the web UI stay up and the camera's settings can be fixed there. Please open a new issue if a wrong camera password still stops motionEye with current releases.

closed 2026-11-08 by JamBalaya56562