motioneye-project/motioneye #272: Time for a user forum?

maglub, 2016-08-05:
Hi,

Perhaps it is time for a .net or .org address and a simple forum for user support?

The issues here are now a mix between user support, feature requests, and bug reports.

A mid term solution is to start using tags for the tickets, so that it is possible to filter (or filter away) issue tickets.

ccrisan, 2016-08-05:
I believe github is just fine for now. A forum would be the same: a mix between bug reports and feature requests, tangled with user support.

I admit labeling issues would make things a lot better and I'll take some time to cleanup the entire issues section.

Regarding a domain on its own - I simply don't have time to setup and manage a forum or any other kind of website specific to motionEye. That's why I prefer doing things here, on github, as almost everything is here, in one place.

maglub, 2016-08-06:
Ok with me! Thanks for the quick answer.

closed 2016-08-06 by maglub