motioneye-project/motioneye #273: Feature request: automated (webhook) recording on/off

Goudtik, 2016-08-05:
I use my smartphone for almost everything. I would like to use a webhook to activate recording when I leave home. I already use the webhook feature for my security system (using IFTTT) Is there a way for you to include a similar option in MotionEye? It would really mean a lot to me! Thanks for the good work so far!

pantchox, 2017-01-01:
did you manage this to work?

barneyc, 2017-08-31:
Just what I am searching for now.  Would really like to be able to start and stop the recording process remotely.  So when we go to bed or leave the house, enable it.  But when we are at home, disable it.  The triggers are simple enough using IFTTT but only if MotionEyes can take an inbound request.

Anyone?

JamBalaya56562, 2026-11-09:
This is the same request as #251 (switch detection on and off from outside the UI), which has the longer discussion, so closing this one as a duplicate. Recent versions have `POST /action/<camera id>/detectionstart` and `detectionpause`; they need the UI's authentication, so a plain webhook cannot call them yet.

closed 2026-11-09 by JamBalaya56562