motioneye-project/motioneye #274: Web-UI -> Motion Notifications Settings Issue

helgetnt, 2016-08-07:
Hallo...
first sorry for my english. I'm from Germany. I've found an issue in the Web-UI -> Motion Notifications Settings. After i type my settings in and press Apply it goes right to the config file (and does the job) but in the Web-UI the options are not in the right line and some are missing. If i make all other changes in the settings before the notifications i've no problem, but if not i have to retype my mail-settings. please have a look at these, because it's quite terrible. 

THX

drcyber, 2017-01-17:
Hi,
first of all thanks for this Nice Software!
I have the same Problem. The testmail works and when i click "Apply" the Settings are moved up one line.
im Runnig it on an RasPi3 with Jessie.

ccrisan, 2017-01-17:
Please try configuring it from scratch. There have been many changes when it comes to configuration since 1 year ago.

drcyber, 2017-01-19:
Thank you , now its working :)

closed 2017-01-19 by ccrisan