motioneye-project/motioneye #275: Contributing

jeff-h, 2016-08-08:
Hey — just wanted to ask if you're open to random feature/bug-fix contributions? I've not seen many projects this popular (especially when you include motionEyeOS) yet with so few contributors.

I wonder if there'd be lots more contributors if this was written in NodeJS :D

Anyway, if you are open to contributions, can I ask, do you have a build process? The files in this repo appear to be just the built files, not the original src? Or do I make changes to these files directly e.g. to change some JavaScript do I change the files inside `static/js/*`?

Thanks for your wonderful work on this over so many years; it's greatly appreciated!

ccrisan, 2016-08-08:
Normally there's nothing to be built for motionEye. Everything, including python and js scripts are interpreted at runtime (on the server and client side, respectively). What are those files that you consider "built" and not original source?

jeff-h, 2016-08-08:
I think my preconceptions gave me the wrong impression. In my day-to-day coding all the JS I write is compiled and concatenated and minified by a build script using Gulp or whatnot — it's refreshing to be able to avoid it all! :)

closed 2016-08-08 by jeff-h

ccrisan, 2016-08-09:
Minifying code is indeed a good practice, especially in the JS world, but given the relatively small JS code that comes with motionEye, minification isn't necessary. Python code is also "minified" or compiled if you want in .pyc files but that happens automatically, at runtime.