motioneye-project/motioneye #276: Webhook connecting the openHAB

GWW88, 2016-08-08:
Gday,

Well done with a wonderful piece of software!

I have been trying to fire a webhook on motion detected and for that to connect to my openHAB instance to trigger a switch. I have had success connecting via IFTTT however I would prefer to go directly from motioneye to openHAB.

The URL I have set up is, https://user:password@192.168.1.5:8443/CMD?MotionEye=ON which works fine when triggered from a browser however does not work when executed from motioneye.

Does motioneye work if the openHAB server has a self signed certificate? 

Regards,
George

ccrisan, 2016-08-08:
That's a good question. It may not work and it may require tweaking. Can you please attach your logs (normally motion.log and motioneye.log)?

GWW88, 2016-08-08:
No worries,
https://dl.dropboxusercontent.com/u/47686770/motion.log
https://dl.dropboxusercontent.com/u/47686770/motioneye.log

Regards,
George

ccrisan, 2016-08-08:
Well your motion.log file doesn't say anything about triggering a webhook. Most likely you have applied some settings recently (and thus motion was restarted) and no motion event has occurred in the meantime to trigger a webhook.

GWW88, 2016-08-08:
Having a look at motioneye.log I thought this may be relevant,

2016-08-02 18:32:37: [motioneye]     INFO: hello! this is motionEye server 20160605
2016-08-02 18:32:39: [motioneye]     INFO: cleanup started
2016-08-02 18:32:39: [motioneye]     INFO: wsswitch started
2016-08-02 18:32:39: [motioneye]     INFO: tasks started
2016-08-02 18:32:39: [motioneye]     INFO: mjpg client garbage collector started
2016-08-02 18:32:39: [motioneye]     INFO: smb mounts started
2016-08-02 18:32:39: [motioneye]     INFO: server started
2016-08-02 18:33:09: [webhook]    ERROR: failed to call webhook: <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:590)>
2016-08-02 18:35:49: [webhook]    ERROR: failed to call webhook: <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:590)>
2016-08-02 18:37:57: [webhook]    ERROR: failed to call webhook: <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:590)>
2016-08-02 18:41:13: [webhook]    ERROR: failed to call webhook: <urlopen error unknown url type: 192.168.1.5>
2016-08-02 18:52:00: [webhook]    ERROR: failed to call webhook: <urlopen error unknown url type: 192.168.1.5>
2016-08-02 18:53:15: [webhook]    ERROR: failed to call webhook: <urlopen error unknown url type: 192.168.1.5>
2016-08-02 18:55:24: [webhook]    ERROR: failed to call webhook: <urlopen error unknown url type: 192.168.1.5>
2016-08-02 18:56:21: [webhook]    ERROR: failed to call webhook: <urlopen error unknown url type: 192.168.1.5>
2016-08-02 18:57:11: [webhook]    ERROR: failed to call webhook: <urlopen error unknown url type: 192.168.1.5>
2016-08-02 18:57:53: [webhook]    ERROR: failed to call webhook: <urlopen error unknown url type: 192.168.1.5>
2016-08-02 18:59:10: [webhook]    ERROR: failed to call webhook: <urlopen error unknown url type: 192.168.1.5>
2016-08-02 19:02:09: [webhook]    ERROR: failed to call webhook: <urlopen error unknown url type: 192.168.1.5>
2016-08-02 19:03:12: [webhook]    ERROR: failed to call webhook: <urlopen error unknown url type: 192.168.1.5>
2016-08-02 19:04:44: [webhook]    ERROR: failed to call webhook: <urlopen error unknown url type: 192.168.1.5>
2016-08-02 19:06:11: [webhook]    ERROR: failed to call webhook: HTTP Error 401: Unauthorized
2016-08-02 19:07:36: [webhook]    ERROR: failed to call webhook: HTTP Error 401: Unauthorized
2016-08-02 19:08:04: [webhook]    ERROR: failed to call webhook: HTTP Error 401: Unauthorized
2016-08-02 19:11:10: [webhook]    ERROR: failed to call webhook: HTTP Error 401: Unauthorized
2016-08-02 19:14:17: [webhook]    ERROR: failed to call webhook: HTTP Error 401: Unauthorized
2016-08-02 19:18:02: [webhook]    ERROR: failed to call webhook: HTTP Error 401: Unauthorized
2016-08-02 19:21:04: [webhook]    ERROR: failed to call webhook: HTTP Error 401: Unauthorized
2016-08-02 19:22:47: [webhook]    ERROR: failed to call webhook: HTTP Error 401: Unauthorized
2016-08-02 19:24:56: [webhook]    ERROR: failed to call webhook: HTTP Error 401: Unauthorized
2016-08-02 19:28:08: [webhook]    ERROR: failed to call webhook: HTTP Error 401: Unauthorized
2016-08-02 19:29:48: [webhook]    ERROR: failed to call webhook: HTTP Error 401: Unauthorized
2016-08-02 19:30:49: [webhook]    ERROR: failed to call webhook: HTTP Error 401: Unauthorized
2016-08-02 19:34:13: [webhook]    ERROR: failed to call webhook: HTTP Error 401: Unauthorized
2016-08-02 19:39:43: [webhook]    ERROR: failed to call webhook: HTTP Error 401: Unauthorized
2016-08-02 19:43:52: [webhook]    ERROR: failed to call webhook: HTTP Error 401: Unauthorized

ccrisan, 2016-08-08:
You're right. My apologies. I forgot I moved the webhook logging to the main motioneye.log file. I'll try to reproduce this issue and get back to you with an answer.

ccrisan, 2016-08-14:
https://github.com/ccrisan/motioneye/commit/d65fcc8980990a79d30a00c74819fc5e8b3a97a5 adds support for unverified HTTPS requests.

closed 2016-08-14 by ccrisan