motioneye-project/motioneye #278: Upload to gdrive does not work

prateeksriv, 2016-08-10:
I am running version 0.33 .
I had admin account configured very early on, so cannot say if that was the trigger.

I have configured google account and generated authentication key.
The pasted the generated key into the configuration field.

I am seeing motion detected and email notifications being sent out but uploading to gdrive does not work at all.

I enabled debug logs and I see prints saying "Code was already redeemed."

`Aug 07 14:43:02 motioneye1 meyectl[2781]: DEBUG: recevied relayed event picture_save for thread id 1 (camera id 1)
Aug 07 14:43:02 motioneye1 meyectl[2781]: DEBUG: adding task "upload_media_file(/var/lib/motioneye/2016-08-07/14-42-55.jpg)" in 5 seconds
Aug 07 14:43:02 motioneye1 meyectl[2781]: DEBUG: saving tasks to "/etc/motioneye/tasks.pickle"...
Aug 07 14:43:02 motioneye1 meyectl[2781]: DEBUG: 200 POST /_relay_event/?_username=admin&event=picture_save&thread_id=1&_signature=something(127.0.0.1) 145.30ms
Aug 07 14:43:03 motioneye1 motion[5780]: [1] [DBG] [ALL] motion_loop: Setup post capture 5
Aug 07 14:43:03 motioneye1 meyectl[2781]: ERROR: gdrive: failed to obtain credentials: Code was already redeemed.
Aug 07 14:43:03 motioneye1 meyectl[2781]: ERROR: failed to upload file "/var/lib/motioneye/2016-08-07/14-42-39.jpg" with service gdrive: Code was already redeemed.
Aug 07 14:43:03 motioneye1 meyectl[2781]: DEBUG: gdrive: uploading file "/var/lib/motioneye/2016-08-07/14-42-40.jpg" to gdrive
Aug 07 14:43:03 motioneye1 meyectl[2781]: DEBUG: gdrive: size of "/var/lib/motioneye/2016-08-07/14-42-40.jpg" is 0.518MB
Aug 07 14:43:03 motioneye1 meyectl[2781]: DEBUG: gdrive: mime type of "/var/lib/motioneye/2016-08-07/14-42-40.jpg" is "image/jpeg"
Aug 07 14:43:03 motioneye1 meyectl[2781]: DEBUG: gdrive: finding folder id for location "/monitoring_189/motioneye1/2016-08-07"
Aug 07 14:43:03 motioneye1 meyectl[2781]: DEBUG: gdrive: requesting credentials
Aug 07 14:43:04 motioneye1 meyectl[2781]: ERROR: gdrive: failed to obtain credentials: Code was already redeemed.
Aug 07 14:43:04 motioneye1 meyectl[2781]: ERROR: failed to upload file "/var/lib/motioneye/2016-08-07/14-42-40.jpg" with service gdrive: Code was already redeemed.`

prateeksriv, 2016-08-10:
I don't understand the problem very well, but I resolved it by following procedure:

1. Get authorization key using the admin control panel ( normal process).
2. Instead of pasting the key into the field in control panel or testing it, I just updated the uploadservices.json file directly.
3. Restarted motion.
4. Checked the debug logs to make sure it worked.

closed 2016-08-10 by prateeksriv

ccrisan, 2016-08-20:
Judging from your error log, it was failing at requesting credentials. Given that the request credentials function depends only on one parameter, the authorization key, I have reasons to believe you tried to configure motionEye's Google Drive uploader using the same authorization key multiple times. It might have partially worked the first time (possibly failing somewhere else) and subsequent requests would have considered the key already "redeemed".