motioneye-project/motioneye #284: wiki information misleading for mr-daves motion 3.4 with ffmpeg 3.0.2 on debian jessie

ozett, 2016-08-17:
hi,
i am in the process of upgrading/fresh install.
i follow the wiki, but there are some misleading information (or i am doing wrong).

i wanted compile the latest motion from mr-dave (which was heavy to find) and is now 3.4.xx

i had no ffmpeg, so i had to installed it from the wiki-description for jessie
(added the repository http://www.deb-multimedia.org  like the wiki said, apt-get update....) 
and got version 3.0.2 .

ffmpeg version N-77463-g4bec36f
libavutil      55. 11.100 / 55. 11.100
  --> looks like ffmpeg 3.0.2 ("Einstein") 3.0.2 was released on 2016-04-28

i had to look this up here https://www.ffmpeg.org/download.html#releases
hard to see the versioning on ffmpeg...

i got netcam_rtsp.c error while compiling motion. is this normal? is it a bug?
can i overcome this?

you gave the hint here https://github.com/ccrisan/motioneye/issues/241#issuecomment-230347143 to use ffmpeg 3.1... but it doesnt seem to be in the wiki-mentioned repo http://www.deb-multimedia.org 

how to overcome? or is the netcam_rtsp.c compile error only cosmetical?

ozett, 2016-08-17:
i wanted to play with converting recordings to .webm format and found vp8 codec missing in ffmpeg from the above repository.

 i looked on ffmpeg page http://johnvansickle.com/ffmpeg/ and found  static-build, whis is version 3.1.2 and has all free codecs in it. only download to a directory..

now i play with this, if i can integrate this version in motion ...

ozett, 2016-08-17:
i got the static ffmpeg package 3.1.2 from ffmpeg site and configured mr-daves motion 3.4.x like this

`./configure --prefix=/usr --without-pgsql --without-sdl --without-sqlite3 --without-mysql --with-ffmpeg= /usr/src/ffmpeg-3.1.2-32bit-static/`

but i got the same netcam_rtsp.c error.

was i wrong with giving the ffmpeg directory as path for configure?
 or are this warnings normal with the rtsp code?

JamBalaya56562, 2026-10-26:
The per-distribution installation pages have been removed from the wiki (December 2025). There is now a single install method for all distributions in the README: https://github.com/motioneye-project/motioneye#installation. `motioneye_init` installs a pre-built motion package and the distribution's ffmpeg, so compiling motion is no longer needed. Closing, as the page this refers to no longer exists. If the README instructions do not work for you, please open a new issue.

closed 2026-10-26 by JamBalaya56562