motioneye-project/motioneye #285: Admin password webui

nathannever, 2016-08-17:
Hello Crison, compliments for your good work. I have a big problem with motioneye webui login, inadvertently I have typed some characters in admin password field and after some other modifications, I have saved settings. Now I cannot login with my password because it was changed. Is there a way to reset admin password? Thank you for your help.

ccrisan, 2016-08-17:
Manually edit it in your motion.conf file.

nathannever, 2016-08-17:
I have put a password in the line @admin_password but it doesn't work.

ccrisan, 2016-08-17:
You must restart the motionEye server.

nathannever, 2016-08-17:
It's a raspberry, I have rebooted but the result is the same.

nathannever, 2016-08-17:
Now it's working! Thank you for your help Crisan. Bye.